rsciriano/ESPHome-OpenTherm #1: OT Not communicating

serhat-ozkara, 2021-03-29:
Hello @rsciriano ,

I've successfully uploaded bin file, but there's no opentherm communication with boiler. May be a pin configuration issue?

I'm using Ihor's adapter
[image]

rsciriano, 2021-03-29:
Hi @serhat-ozkara, the pins configuratios is in [opentherm_component.h](https://github.com/rsciriano/ESPHome-OpenTherm/blob/f1bd6ee2ff7bc2eb8b44fb194972a9d50684a39b/opentherm_component.h#L11), review this and change it with your values

rsciriano, 2021-03-29:
@serhat-ozkara, also you can use the current pins

```
const int inPin = D5; 
const int outPin = D6; 
```

serhat-ozkara, 2021-03-29:
@rsciriano Hello,

I managed to get it working by defining 

`const int inPin = D2; 
 const int outPin = D1;`

Now opentherm works!
But OT values from boiler seems to be not working:
[image]

rsciriano, 2021-03-29:
@serhat-ozkara try to change the constructor of [`OpenthermComponent`](https://github.com/rsciriano/ESPHome-OpenTherm/blob/f1bd6ee2ff7bc2eb8b44fb194972a9d50684a39b/opentherm_component.h#L37) class

from 
```
  OpenthermComponent(): PollingComponent(15000) {

  }
```

to 
```
  OpenthermComponent(): PollingComponent(3000) {

  }
```

serhat-ozkara, 2021-03-29:
@rsciriano  Actually, I noticed this on opentherm_component.h:

`// Read sensor values    
    /*
    float boilerTemperature = ot.getBoilerTemperature();
    float ext_temperature = getExternalTemperature();
    float pressure = getPressure();
    float modulation = getModulation();

    // Publish sensor values
    //flame->publish_state(isFlameOn); 
    external_temperature_sensor->publish_state(ext_temperature);
    return_temperature_sensor->publish_state(return_temperature);
    boiler_temperature->publish_state(boilerTemperature);
    pressure_sensor->publish_state(pressure);
    modulation_sensor->publish_state(modulation);
    */`
    
    They're all commented-out :)

rsciriano, 2021-03-29:
> @rsciriano Actually, I noticed this on opentherm_component.h:
>
> `// Read sensor values
> /*
> float boilerTemperature = ot.getBoilerTemperature();
> float ext_temperature = getExternalTemperature();
> float [...]

Yes, I commented that code because I did the ones you can do without the opentherm adapter. The important thing is the heating climate and return temperature values.

Aliskin-Papa, 2021-04-22:
Switch "Termostato ambiante" -> calculate Target temperature without PID.

closed 2021-06-23 by rsciriano