rsciriano/ESPHome-OpenTherm #2: Hot Water Thermostat not show status Work

Aliskin-Papa, 2021-04-22:
Hot Water Thermostat good work, but not change mode in the lovelas card to Work, any time show status Not Work.

rsciriano, 2021-04-22:
Hi @Aliskin-Papa, in my boiler works. When I turn on the tap, the status changes

[image]

The status is retrieved using the `isHotWaterActive` method of OpenTherm library, it is possible that your boiler does not return this information.

```
bool isHotWaterActive = ot.isHotWaterActive(response);
```

Aliskin-Papa, 2021-04-23:
My boiler, Viessman Vitopend 100, can't return Pressure, Return temp and HotWaterActive. Its bad news for me...
Thanks to your work, its brilliant!

Aliskin-Papa, 2021-04-23:
And one small question - how work switch Temperature ambiento?

closed 2021-04-23 by Aliskin-Papa