rsciriano/ESPHome-OpenTherm #4: There have been 10 suspected unsuccessful boot attempts.

sag3ll0, 2021-07-06:
after flash i got this in logs view
i using wemos d1 mini and esphome flasher
`Hard Resetting...
Done! Flashing is complete!

Showing logs:
[22:48:11][I][logger:170]: Log initialized
[22:48:11][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[22:48:11]
[22:48:11]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[22:48:11]
[22:48:11]Exception (28):
[22:48:11]epc1=0x402170fb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000024 depc=0x00000000
[22:48:11]
[22:48:11]>>>stack>>>
[22:48:11]
[22:48:11]ctx: cont
[22:48:11]sp: 3ffffc70 end: 3fffffc0 offset: 0190
[22:48:11]3ffffe00:  3fff19ac 40217f9c 00000020 401011ee  
[22:48:11]3ffffe10:  3fff19ac 3ffffed0 3fff43b4 4024cf0e  
[22:48:11]3ffffe20:  3ffffe01 3fff2dac 00000001 4022a52c  
[22:48:11]3ffffe30:  3fff3e14 3fff1958 00000000 40217fb8  
[22:48:11]3ffffe40:  3fff4070 00000000 00000000 3fff2dac  
[22:48:11]3ffffe50:  00000000 00000000 00000000 4022fce0  
[22:48:11]3ffffe60:  00000000 4022fce0 00000000 4022fce0  
[22:48:11]3ffffe70:  00000000 4022fce0 00000000 4022fce0  
[22:48:11]3ffffe80:  00000000 00000000 3fff3fe8 00000000  
[22:48:11]3ffffe90:  00000000 3fff4000 00000000 00000000  
[22:48:11]3ffffea0:  00000000 4022fce0 00000000 4022fce0  
[22:48:11]3ffffeb0:  00000000 4022fce0 00000000 4022fce0  
[22:48:11]3ffffec0:  00000000 4022fce0 00000000 00000000  
[22:48:11]3ffffed0:  3fff3e0c 0073746e 402167b8 402170e8  
[22:48:11]3ffffee0:  feefeffe feefeffe feefeffe feefeffe  
[22:48:11]3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff00:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff10:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff20:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff30:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff40:  feefeffe feefeffe feefeffe feefeffe  
[22:48:12]3fffff50:  feefeffe feefeffe feefeffe 3fff43b4  
[22:48:12]3fffff60:  3fff427c feefeffe feefeffe feefeffe  
[22:48:12]3fffff70:  feefeffe feefeffe feefeffe 3fff4410  
[22:48:12]3fffff80:  3fff43b4 feefeffe feefeffe feefeffe  
[22:48:12]3fffff90:  feefeffe feefeffe feefeffe 3fff1d54  
[22:48:12]3fffffa0:  3fffdad0 00000000 3fff1d14 4022a9b8  
[22:48:12]3fffffb0:  feefeffe feefeffe 3ffe8640 40100731  
[22:48:12]<<<stack<<<
[22:48:12]
[22:48:12]--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

rsciriano, 2021-07-06:
Hi @sag3ll0,

You can try the latest version of the code I just published (thanks to the contribution of @fwestenberg)

If it still doesn't work, you would use [Command-line Decoder for Strack Trace from ESP8266](https://github.com/littleyoda/EspStackTraceDecoder) (or the [Arduino IDE version](https://github.com/me-no-dev/EspExceptionDecoder)) to try to reach the code where the error occurs.

sag3ll0, 2021-07-07:
I have already tried these changes, the result is the same. It is important to add that I tried the code without connecting to the heater. Perhaps this is the cause of the problems. But this is strange in theory, if there is no data from the opentherm bus, then the application should not crash. In general, when I try with the boiler, everything will stand up clearly.

rsciriano, 2021-07-08:
Hi @sag3ll0, 

The code needs a lot of improvements, one of them is this one.

In this case it should continue to work and simply report that there is a connection problem on the opentherm bus.

I have modified the issue title to make it more clear :-)

MaartenVerb, 2022-02-08:
Hello, I also ran into this problem. I first wanted to flash, check if everything looked ok and then replace my current on/off thermostat module. 

Now I was not sure if the code was ok, because the whole ESP8266 hangs. 

If this is an issue that can be solved, that would be nice, it will save the next guy half a day of troubleshooting haha.

foxalex85, 2022-09-10:
Hello, there are news about this issue? I've flashed my esp and I would to test the thermostat without connect to Opentherm bus, but it's won't start. 
Thanks

MaartenVerb, 2022-09-14:
This is still a big issue. My Remeha HR heater is compatible with on/off and openTherm. But if the heater does not detect opentherm, it keeps on/off mode. The module crashes as soon as it is not connected. It should wait and retry every 10 seconds or something.

Please advice!

MaartenVerb, 2023-10-17:
@rsciriano Hey Robert, This is a blocking issue for me, do you have a workaround or planning when this could be fixed? Kind regards.

MaartenVerb, 2023-10-20:
Yea, confirmed, this issue can be closed.