rsciriano/ESPHome-OpenTherm #5: HA - ESPhome config

marcos-acpc, 2021-07-08:
Hi I managed to flash it but when I go to add it on ESPHome it keeps giving me an API error.
[image]

ps: to flash got needed to coment this line on opentherm_climate:  'traits.set_supports_heat_cool_mode(false);'

marcos-acpc, 2021-07-08:
[image]
print of the code I used from you

rsciriano, 2021-07-08:
Hi @marcos-acpc, 
I think you are using a old version of esphome.

Try to upgrade esphome, uncomment the line with `traits.set_supports_heat_cool_mode` and compile again.

marcos-acpc, 2021-07-08:
Hi thanks for the fast response. I will update when I return from my holidays. Does this line of code influence the config on Esphome server?  Sorry but I didn't got time to go thru all code again and just just start reading and understanding it. Thanks

rsciriano, 2021-07-08:
Hi @marcos-acpc,

The compilation error that you got in the method `traits.set_supports_heat_cool_mode` was caused because, in recent versions, they have introduced a breaking change (they have renamed this method from `set_supports_auto_mode`).

This has indicated to me that you were using an older version of esphome and that this is the cause of the communication error with Home Assistant :-)

marcos-acpc, 2021-07-08:
Hi, thank you for the explanation I will check next available opportunity.

whxciotw, 2021-07-15:
@rsciriano Hi, I face the similar problem.
The version of esphome is 1.19.4, and the version of HA is 2021.6.6.

When I compiling the firmware, esphome shows the log as below:
`Compiling .pioenvs/opentherm/src/main.cpp.o
In file included from src/opentherm_component.h:5:0,
                 from src/main.cpp:40:
src/opentherm_climate.h: In member function 'virtual esphome::climate::ClimateTraits OpenthermClimate::traits()':
src/opentherm_climate.h:46:68: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:60): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_heat_cool_mode(supports_heat_cool_mode_);
                                                                    ^
src/opentherm_climate.h:48:44: warning: 'void esphome::climate::ClimateTraits::set_supports_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:56): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_cool_mode(false);
                                            ^
src/opentherm_climate.h:49:43: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:58): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_heat_mode(true);
                                           ^
src/opentherm_climate.h:51:39: warning: 'void esphome::climate::ClimateTraits::set_supports_away(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:119): This method is deprecated, use set_supported_presets() instead [-Wdeprecated-declarations]
         traits.set_supports_away(false);
                                       ^
Linking .pioenvs/opentherm/firmware.elf
Retrieving maximum program size .pioenvs/opentherm/firmware.elf
Checking size .pioenvs/opentherm/firmware.elf
Building .pioenvs/opentherm/firmware.bin`

After flashing the firmware to the esp board, the esp cannot communicate to the HA or Esphome, it seems that the esp board can connect to wifi, but cannot build connection to HA server or Esphome server via port 6053.
Besides, the esp board can show the log via usb2ttl, and below is the log via usb2ttl.
`Showing logs:
[22:07:15]{l[I][logger:170]: Log initialized
[22:07:15][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[22:07:15][W][app:022]: Component already registered! (0x3fff3d9c)
[22:07:15][I][app:029]: Running through setup()...
[22:07:15][D][opentherm_component:044]: Setup
[22:07:16][D][sensor:117]: 'PID Climate Result': Sending state 0.00000 % with 1 decimals of accuracy
[22:07:16][D][sensor:117]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[22:07:16][D][sensor:117]: 'PID Climate ERROR': Sending state 0.00000 % with 1 decimals of accuracy
[22:07:16][C][wifi:037]: Setting up WiFi...
[22:07:16][I][wifi:235]: WiFi Connecting to 'Calmdown'...
[22:07:16][D][opentherm_component:095]: update heatingWaterClimate: 0
[22:07:16][D][opentherm_component:096]: update hotWaterClimate: 0
[22:07:19][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)
[22:07:26][D][binary_sensor:036]: 'Flame': Sending state OFF
[22:07:26][D][sensor:117]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[22:07:26][D][sensor:117]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[22:07:26][D][sensor:117]: 'Boiler Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[22:07:26][D][sensor:117]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[22:07:26][D][sensor:117]: 'Boiler Modulation': Sending state -1.00000 % with 0 decimals of accuracy
[22:07:26][D][sensor:117]: 'Heating Target Temperature': Sending state 10.00000 ºC with 1 decimals of accuracy
[22:07:26][D][climate:369]: 'Hot water' - Sending state:
[22:07:26][D][climate:372]:   Mode: OFF
[22:07:26][D][climate:374]:   Action: OFF
[22:07:26][D][climate:392]:   Current Temperature: -1.00°C
[22:07:26][D][climate:398]:   Target Temperature: 0.00°C
[22:07:26][D][climate:369]: 'Heating water' - Sending state:
[22:07:26][D][climate:372]:   Mode: OFF
[22:07:26][D][climate:374]:   Action: OFF
[22:07:26][D][climate:392]:   Current Temperature: 0.00°C
[22:07:26][D][climate:396]:   Target Temperature: Low: 0.00°C High: 0.00°C
[22:07:26][I][wifi:498]: WiFi Connected!
[22:07:26][C][wifi:344]:   SSID: 'Calmdown'
[22:07:26][C][wifi:345]:   IP Address: 192.168.19.13
[22:07:26][C][wifi:347]:   BSSID: FC:7C:02:BB:83:28
[22:07:26][C][wifi:348]:   Hostname: 'opentherm'
[22:07:26][C][wifi:352]:   Signal strength: -33 dB ▂▄▆█
[22:07:26][C][wifi:356]:   Channel: 6
[22:07:26][C][wifi:357]:   Subnet: 255.255.255.0
[22:07:26][C][wifi:358]:   Gateway: 192.168.19.1
[22:07:26][C][wifi:359]:   DNS1: 192.168.19.1
[22:07:26][C][wifi:360]:   DNS2: (IP unset)
[22:07:26][D][wifi:507]: Disabling AP...
[... 144 lines cut ...]
[22:08:26][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)
[22:08:32][D][sensor:117]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[22:08:32][D][sensor:117]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[... cut ...]
[22:08:32][C][pid.sensor:056]: PID Climate Sensor 'PID Climate Result'
[22:08:32][C][pid.sensor:056]:   State Class: 'measurement'
[22:08:32][C][pid.sensor:056]:   Unit of Measurement: '%'
[22:08:32][C][pid.sensor:056]:   Accuracy Decimals: 1
[22:08:32][C][pid.sensor:056]:   Icon: 'mdi:gauge'
[22:08:32][D][opentherm_component:095]: update heatingWaterClimate: 0
[22:08:32][D][opentherm_component:096]: update hotWaterClimate: 0
[22:08:35][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)`

marcos-acpc, 2021-07-18:
> Hi @marcos-acpc,
> I think you are using a old version of esphome.
> 
> Try to upgrade esphome, uncomment the line with `traits.set_supports_heat_cool_mode` and compile again.

Hi I just returned, I updated the esphome and  managed to flash it with your original file. It connects to network fine but the error on home assistant persists.

marcos-acpc, 2021-07-18:
to add this is showing on when flashing:
[image]

marcos-acpc, 2021-07-19:
I found this on esphome documentation but I don't understand what needs to be changed to make it work:
https://esphome.io/api/classesphome_1_1climate_1_1_climate_traits.html

marcos-acpc, 2021-07-20:
@rsciriano I tried to change it and make it work but for now, is out of reach for me as I don't know much about c++.

marcos-acpc, 2021-07-20:
make it to fully connect only with it attached to the boiler.. water pressure dont work but it might be from boiler.

rsciriano, 2021-07-20:
Hi @marcos-acpc,

I have been testing and the connection error occurs with version 2021.6.6 of Home Assistant. 

Upgrading to version 2021.7.3 solves the problem :-)

rsciriano, 2021-07-20:
> make it to fully connect only with it attached to the boiler.. water pressure dont work but it might be from boiler.

Yes, some boilers provide less information than the opentherm protocol supports. _Return Temperature_ and  _External Temperature_ are often missing

marcos-acpc, 2021-07-20:
@rsciriano but the external sensor you refer to on the read me the file is supposed to be inside or outside the house?

rsciriano, 2021-07-20:
> @rsciriano but the external sensor you refer to on the read me the file is supposed to be inside or outside the house?

Yes, The external sensor I refer to on readme is inside the house. That sensor is used in the thermostat (PID Climate)

In my case, this sensor are in another device, in another room. To use it, need declare on YAML as a Home Assistant sensor 
(line 82)

```yaml
  - platform: homeassistant
    id: temperature_sensor
    entity_id: sensor.temperature_sensor
    name: "Actual temperature"
    unit_of_measurement: °C
```

- The `id` property value have been always`temperature_sensor`
- The `entity_id` property value is the **Entity Id** of the sensor on Home Assistant

---

_On the other hand,
The sensor I was referring to before, is another one that can be connected to the boiler and read via opentherm._

marcos-acpc, 2021-07-20:
> Yes, The external sensor I refer to on readme is inside the house. That sensor is used in the thermostat (PID Climate)
>
> In my case, this sensor are in another device, in another room. To use it, [...]

I got this. it was just to double check. :) The problem with the connection with HA, gone when I plug it into the boiler. **Only after was ON and running with the boiler**, that I managed to connect on esphome integrations. 
I would like to add but if possible could you please add some explanation on this:
[image]
I understand what they ruffly means but I don't understand the percentages and the autotune. With this everyone would understand better what to look for.

Johnssj3, 2021-08-28:
I have the same issue ESPhome will not connect 
[image]

ananyevgv, 2021-09-07:
release v0.2 everything works, release v0.3 does not work 
core-2021.9.4
supervisor-2021.09.0
ESPHome version: 2021.8.2

ananyevgv, 2021-09-10:
First you need to connect the boiler, then it works. If the boiler is not connected, the API is not connected v.0.3

ShurikenGitHub, 2021-10-31:
> 
> 
> I found this on esphome documentation but I don't understand what needs to be changed to make it work: https://esphome.io/api/classesphome_1_1climate_1_1_climate_traits.html

I fixed the warnings, look at my pull request/fork,
If you want to add another climatemode, just add a comma and the one you like from this list.
[https://esphome.io/api/namespaceesphome_1_1climate.html#a8716cd7ed01ceb7ccedfa2f7b47a39d2](url)

marcos-acpc, 2021-10-31:
@ShurikenGitHub  Hi, thanks for the contribution. The only thing I need to test is when I turn off the thermostat the pumps keep running as normal but the flame turns on for few seconds time to time and keeps my house at 21c degrees but ye some testing will be required during this winter as gas prices are getting too high. And I do not want to turn on the heating till I really need it.

marcos-acpc, 2021-11-24:
@ShurikenGitHub I was trying to update my esp device but keep getting this: 
[image]
seems some error with dependencies but I dont know how to fix it :(

rsciriano, 2021-11-25:
Hi @marcos-acpc, I think the problem could be with the version of esphome Home Assistant add-on (a similar issue has reported [here](https://github.com/esphome/issues/issues/643))

The issue could be solved by updating the add-on

ShurikenGitHub, 2021-11-25:
> Hi @marcos-acpc, I think the problem could be with the version of esphome Home Assistant add-on (a similar issue has reported [here](https://github.com/esphome/issues/issues/643))
> 
> The issue could be solved by updating the add-on

I'm running the latest version of the esphome addon and I get the same error.

rsciriano, 2021-11-25:
@ShurikenGitHub Yes, Could it be the Arduino framework version. Try commenting out or removing the [9 line](https://github.com/rsciriano/ESPHome-OpenTherm/blob/22353c087758b580370861758dc7e29a5e0f6dbb/opentherm.yaml#L9) on `opentherm.yam` file with

```yaml
arduino_version: '2.7.2'  
```

marcos-acpc, 2021-11-25:
I comented the Arduino  and the esp async we server lines and now it compiles fine I will check all the functions later on.

HelpITru, 2022-01-05:
> I comented the Arduino and the esp async we server lines and now it compiles fine I will check all the functions later on.

Everything OK ? I have some problem with this code.

marcos-acpc, 2022-01-05:
So far so good with me

megavolt163, 2022-05-13:
Can you tell me? If my ESP is not connected to the boiler, then I will not get entities in esphome?

[image]

megavolt163, 2022-05-13:
@rsciriano,  can you help me? 
At logs:
[20:44:34][D][climate:408]:   Current Temperature: -1.00°C
[20:44:34][D][climate:414]:   Target Temperature: 0.00°C
[20:44:34][D][climate:385]: 'Heating water' - Sending state:
[20:44:34][D][climate:388]:   Mode: OFF
[20:44:34][D][climate:390]:   Action: OFF
[20:44:34][D][climate:408]:   Current Temperature: 0.00°C
[20:44:34][D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[20:44:35][D][api:102]: Accepted 192.168.1.199
[20:44:35][W][api.connection:083]: Home Assistant 2022.5.4 (192.168.1.199): Connection closed

File  opentherm_component.h have^
  OpenthermComponent(): PollingComponent(10000) {

but in Home Assistant i have not sensor.