rsciriano/ESPHome-OpenTherm #6: Temperature on DeDietrich Bolier

macgyverek, 2021-08-01:
Hi,

Today I install your code to ESP8266 and connect to the boiler. There is any option to send current room temperature to the boiler via OpenTherm standard? 
On DeDietrich screen I have message that CIRCA controlled via Opentherm module but I have "-----" in place where should be current room temperature.  

THX for your work!

marcos-acpc, 2021-08-02:
Hi did you add your external temperature into the code?

macgyverek, 2021-08-02:
Yes, I added. Is working on Home Assistant. But this temperature is not sending back to DeDietrich Boiler.

marcos-acpc, 2021-08-02:
That seems odd but when I tested it worked for me. Only the PID version is a bit harder as I don't know what it all means and do.  I have an ideal logic+ so I can't help much as boilers implement open therm differently. try different sensors can u see any reading on the actual temperature sensor.

rsciriano, 2021-08-02:
> Hi,
>
> Today I install your code to ESP8266 and connect to the boiler. There is any option to send current room temperature to the boiler via OpenTherm standard?
> On DeDietrich screen I have message [...]

Hi @macgyverek,

The current room temperature can be sent to the boiler, but several code changes are necessary to achieve this.

I have made these changes in [this new branch](https://github.com/rsciriano/ESPHome-OpenTherm/tree/room_current_temperature), and you can see them in the [branch comparison](https://github.com/rsciriano/ESPHome-OpenTherm/compare/room_current_temperature)

I haven't tried them connected to the boiler but they should work :-)

macgyverek, 2021-08-03:
This version not working. Many connection and disconnection from Home Assistant. MQTT not see data.
On logs is: connecting to Home Assistant and Dissconnecting from Assistant. Hmmm.

ananyevgv, 2021-08-13:
good afternoon, the boiler does not work for me either, the following errors occur when compiling

ananyevgv, 2021-08-13:
In file included from src/main.cpp:60:0: src/opentherm_climate.h: In member function 'virtual esphome::climate::ClimateTraits OpenthermClimate::traits()': src/opentherm_climate.h:47:49: warning: 'void 
esphome::climate::ClimateTraits::set_supports_heat_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:60): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations] 
traits.set_supports_heat_cool_mode(false); ^ src/opentherm_climate.h:48:44: warning: 'void esphome::climate::ClimateTraits::set_supports_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:56): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations] 

traits.set_supports_cool_mode(false); ^ src/opentherm_climate.h:49:43: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:58): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations] 

traits.set_supports_heat_mode(true); ^ src/opentherm_climate.h:51:39: warning: 'void esphome::climate::ClimateTraits::set_supports_away(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:119): This method is deprecated, use set_supported_presets() instead [-Wdeprecated-declarations] traits.set_supports_away(false);