rsciriano/ESPHome-OpenTherm #10: Boiler setup question

AndreyShpilevoy, 2021-09-22:
Hi, I have a question about the correct setup.
I have a boiler Remeha Tzerra Ace 39c with eTwist thermostat.
And I would like to control the boiler thru Home Assistant next to eTwist thermostat.

I bought [this adapter](https://diyless.com/product/esp8266-opentherm-gateway) and checked example [this](https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermGatewayMonitor_Demo/OpenThermGatewayMonitor_Demo.ino).
As far as I understand we need 4 pins to communicate. 2 for master mode and 2 for slave mode.
Is it possible to connect this device with your ESPHome firmware to the boiler and still have the possibility to control it via eTwist thermostat in case if needed? Or it's designed to work only in Master mode and replace the thermostat at all?

rsciriano, 2021-09-22:
Hi @AndreyShpilevoy, 

My FW does not support this behaviour, but @wanno-drijfhout has created a fork where he is working just on this, you can see his progress [here](https://github.com/wanno-drijfhout/ESPHome-OpenTherm/pull/1) :-)

AndreyShpilevoy, 2021-09-22:
That's great news!
I will be tuned for it.
@wanno-drijfhout in case if you will need some testing after all.

ananyevgv, 2021-09-30:
https://github.com/wichers/esphome-opentherm

wanno-drijfhout, 2021-10-03:
> https://github.com/wichers/esphome-opentherm

Thanks for sharing, @ananyevgv ! That codebase definitely solved a few issues I was already struggling with and provides some new ideas. I also tried flashing that codebase on my Gateway. I probably did it wrong because my boiler just turned on at maximum capacity -- the behavior that occurs if communication with the thermostat doesn't work.

Perhaps @AndreyShpilevoy would like to try wichers' codebase to see if it works?