rsciriano/ESPHome-OpenTherm #11: hot water BAXI

ananyevgv, 2021-09-25:
I can't set the temperature of hot water, the hot water heating mode is turned on but the water is not heated

marcos-acpc, 2021-09-28:
Hi do you have pid_climate_controller on climate card with the target temperature?

ananyevgv, 2021-09-30:
Good afternoon. Yes, there is, but I turned it off, and if I understood correctly, it is responsible for heating water, not hot water

rsciriano, 2021-10-01:
> Good afternoon. Yes, there is, but I turned it off, and if I understood correctly, it is responsible for heating water, not hot water

Hi @ananyevgv, what you say is right, the `pid_climate_controller` is only for heating water.  

Hot water is only controlled by the **Hot water climate**. I think it could be a communication problem with the boiler.

I have read in another issue that you have a BAXI nuvola 3 comfort 240 fi. According to [this compatibility matrix](https://otgw.tclcode.com/matrix.cgi), your boiler is compatible with the message that sets the hot water temperature, I don't know what might be happening. 

[image]

I recommend you try the most basic code, directly with the [Ihor Melnyk examples](http://ihormelnyk.com/opentherm_library), and go step by step. 

I'm sorry I can't be of more help

P.D: I have a Baxi Platinum Compact ECO, and it works fine

ananyevgv, 2021-10-04:
Hello @rsciriano ,
Thank you so much for your work. I will figure it out further, there are two ideas what else can be done, I will try. Unfortunately, it's cold time now, and I don't really want to experiment with the boiler. Thanks again.

ananyevgv, 2021-10-04:
In one of the descriptions of the ready-made solution with opentherm (https://zont-online.ru ) found a mention of :
"Second Circuit
The option necessary to turn on the DHW control on
some boilers" 
and we are talking just about BAXI

ananyevgv, 2021-10-05:
Hello @rsciriano ,
I found it on the Internet "for proper communication with this boiler, it is advisable to ask him the Slave configuration (MSGID=2) first, answer him with the same code in the Master configuration (MSGID=3), and then send the Slave product version (MSGID=127)"
    MSGID=2	MConfigMMemberIDcode, // flag8 / u8  Master Configuration Flags /  Master MemberID Code
    MSGID=3    SConfigSMemberIDcode, // flag8 / u8  Slave Configuration Flags /  Slave MemberID Code
    MSGID=127   SlaveVersion, // u8 / u8  Slave product version number and type

ananyevgv, 2021-11-15:
I found it on the Internet, "For absolutely correct communication with this boiler, it is advisable to ask him the Slave configuration (MSGID=2) first, answer him with the same code in the Master configuration (MSGID=3), and then send the Slave product version (MSGID=127)."

ananyevgv, 2021-11-15:
MsgID=127 «Slave product version: 51 1», т.е. [1 000 0000 01111111 00000000 00000000],

michalrychlik, 2021-12-25:
Hi, I have the exact same issue. Did you figure out how to solve it?

ananyevgv, 2021-12-27:
I decided to do it differently, since this option does not implement the opentherm protocol in full. I decided to do it with a controller and a serial-to-wifi bridge. (links below. unfortunately, only in Russian language).

https://modkam.ru/2018/09/13/shljuz-opentherm-versija-2/
https://modkam.ru/2017/12/14/shljuz-opentherm/

https://github.com/oxan/esphome-stream-server

I want to do it all in the watch case. Unfortunately, it is very cold now (-20), so I left the experiments with the boiler until the summer.
The appearance and 3D model are here:
https://github.com/ananyevgv/Esphome-clock-NTP/tree/main/clock-3

ananyevgv, 2022-03-10:
I drew a board for your program, I can upload files for the order

ananyevgv, 2022-03-10:
https://github.com/ananyevgv/shild/blob/main/shild%20OpenthermESP32%2BAC-Melnik.zip

ananyevgv, 2022-08-25:
Hello @rsciriano , 
I've tried everything, your code is the best, I found it on the Internet:

      unsigned long request3 = ot.buildRequest(OpenThermRequestType::READ, OpenThermMessageID::SConfigSMemberIDcode, 0xFFFF);
      unsigned long respons3 = ot.sendRequest(request3);
      uint8_t SlaveMemberIDcode = respons3 >> 0 & 0xFF;
      unsigned long request2 = ot.buildRequest(OpenThermRequestType::WRITE, OpenThermMessageID::MConfigMMemberIDcode, SlaveMemberIDcode);
      unsigned long respons2 = ot.sendRequest(request2);

      unsigned long request127 = ot.buildRequest(OpenThermRequestType::READ, OpenThermMessageID::SlaveVersion, 0);
      unsigned long respons127 = ot.sendRequest(request127);
      uint16_t dataValue127_type = respons127 & 0xFFFF;
      uint8_t dataValue127_num = respons127 & 0xFF;
      unsigned result127_type = dataValue127_type / 256;
      unsigned result127_num = dataValue127_num;
      unsigned long request126 = ot.buildRequest(OpenThermRequestType::WRITE, OpenThermMessageID::MasterVersion, 0x013F);
      unsigned long respons126 = ot.sendRequest(request126);
      uint16_t dataValue126_type = respons126 & 0xFFFF;
      uint8_t dataValue126_num = respons126 & 0xFF;
      
      unsigned result126_type = dataValue126_type / 256;
      unsigned result126_num = dataValue126_num;
      Serial.println("Boiler type and version: type " + String(result126_type) + ", ver " + String(result126_num));
      Serial.println("Boiler type and version: type " + String(result127_type) + ", ver " + String(result127_num));

 localRequest = ot.buildRequest(OpenThermRequestType::READ, OpenThermMessageID::SConfigSMemberIDcode, 0xFFFF);
      sendRequest(localRequest, localResponse);

      if(!vars.monitor_only.value && ot.getLastResponseStatus() == OpenThermResponseStatus::SUCCESS && ot.getDataID(localResponse) == OpenThermMessageID::SConfigSMemberIDcode) 
      {
      localRequest = ot.buildRequest(OpenThermRequestType::WRITE, OpenThermMessageID::MConfigMMemberIDcode, vars.SlaveMemberIDcode.value);
      sendRequest(localRequest, localResponse);
      }
      
      localRequest = ot.buildRequest(OpenThermRequestType::READ, OpenThermMessageID::SlaveVersion, 0);
      sendRequest(localRequest, localResponse);

      localRequest = ot.buildRequest(OpenThermRequestType::WRITE, OpenThermMessageID::MasterVersion, 0x013F);
      sendRequest(localRequest, localResponse);

      DEBUG.println("Boiler type and version: type" + String(vars.MasterType.value) + ", ver " + String(vars.MasterVersion.value));
      DEBUG.println("Boiler type and version: type " + String(vars.SlaveType.value) + ", ver " + String(vars.SlaveVersion.value));

ananyevgv, 2023-05-12:
It turned out, a little later I will post the code for baxi nuvola-3 b40

ananyevgv, 2023-05-13:
https://github.com/ananyevgv/ESPHome-OpenTherm-Baxi_nuvola/tree/main/esphome-opentherm

closed 2023-05-13 by ananyevgv