rsciriano/ESPHome-OpenTherm #13: Heating water Viessman 100w combi boiler

trueskalon, 2021-09-30:
Hi,

maybe someone will help me or will give me some hints.

Iam using OT gatway and rsciriano ESPHome-OT with Viessman 100w combi boiler.

Ive tried v0.1 to v0.3.  

Iam using node-red and call service nodes.

I can feed set_temperature via climate.heating_water and climate.hot_water.

Hot water is activeted by water flow sensor in the boiler (water heating is staring).

I got issue with heating water. 

Whaen I call servic - turn_on via climate.heating_water only  circulation water pomp is starting. 

Water is not being heated.

Guys maybe someone could help me with getting it work.

Thanks.

ananyevgv, 2021-10-01:
P07 
0" - OpenTherm is activated
Wi-Fi enabled full access
"1" - OpenTherm is activated only as a thermostat
Wi-Fi enabled full access
"2" - OpenTherm is deactivated.
Wi-Fi activated full access
"3" - OpenTherm activated
Wi-Fi is deactivated.
"4" - OpenTherm is activated only as a thermostat
Wi-Fi is deactivated.

trueskalon, 2021-10-01:
Thanks, my boiler is from 2015 so its a model with knobs, but I will check that what Ive got on service mode on my boiler.

rsciriano, 2021-10-01:
Hi @trueskalon, 

The boiler has different connectors for on/off thermostat and opentherm, , you have to make sure that you have connected it to the correct one in the picture below, connector B)

[image]

To simplify testing, I recommend using v0.1 (the PID is more complicated to configure).

Also, you have to check the configuration of the pins connected to the ESP, in this version they are: 

- inPin = D5 
- outPin = D6 

By the way, the old thermostat must be disconnected. I see you have a gateway, my code is prepared for the [Ihor Melnyk adapter](http://ihormelnyk.com/opentherm_adapter), in theory the gateway is two adapters, but you have to be careful with its configuration.

Good luck

ananyevgv, 2021-10-01:
Hi @trueskalon, I have a similar situation with BAXI nuvola 3 comfort 240 fi , while looking for solutions. While I'm inclined that BAXI implementation of opentherm is different, according to your boiler, it came across in the forums that you need to change the parameter P07.

rsciriano, 2021-10-01:
> Thank you, I done all that you have pointed out - Ihor gateway, esp - everything is setup and working.
> 
> My issue is that when I use call service node, climate.heating.water - turn_on the circulation pump is starting but flame is off (before that I feed via call service node my boiler with heating water temperature). The boiler is in heat mode.

Ok @trueskalon, now I understand what might be happening.

The `climate.heating.water` is only to turn on the heatting and set the heatting watter temperature. For the flame to ignite, it is necessary to turn on `switch.thermostat_ambiente` (sorry for the spanish name), I used it with the generic thermostat on Home Assistant. 

> Important!!!, 
> This only works with the v0.1, the other versions are controlled by the PID thermostat.

trueskalon, 2021-10-02:
Hello, still no luck...

Switching thermostat "on" and setting heating water "on" does not turn on the flame in the boiler :(.

I double checked everything, reed manual for my Vitodens 100w (its old model B1HA with no service OT modes like in new ones).

In the esp logs, I can see that "action" for heating water is "off" and this seems to be an issue. I do not know how to make action for heating water in state "on".

[image]