rsciriano/ESPHome-OpenTherm #17: Hot water current temperature is 0

enver-ablyaev, 2021-12-09:
In HA hot water current temp is 0 while I can see the correct temperature from the service menu on my boiler (Buderus Logamax U072)
Please advise how to fix it
[image]

IvAn-source-git, 2022-01-14:
Hi. Do you have a single or double boiler? For single-loop, you need to edit the library OpenTherm.h
	Tr, // f8.8  Room temperature (°C)
	Tboiler, // f8.8  Boiler flow water temperature (°C)
	Tstorage, // f8.8  Solar storage temperature (°C)
	Toutside, // f8.8  Outside temperature (°C)
	Tret, // f8.8  Return water temperature (°C)
	Tdhw, // f8.8  DHW temperature (°C)

enver-ablyaev, 2022-01-14:
@IvAn-source-git where this file is located? 
I can't find it in ESPHome-OpenTherm folder. 
Please advise.

enver-ablyaev, 2022-01-14:
ok, from your repo I understood I simply can download it to my RPI and change it there.
but what exactly do I have to change there?

IvAn-source-git, 2022-01-14:
Take everything from my repo. A lot has been fixed for our boiler (U)72 and Gaz60000)

bartmarchand1, 2022-01-19:
@IvAn-source-git  what do you mean with a single or double loop?
I have a Bulex Thermomaster 28F, for heating and hot tap water production. Is this the double loop?

IvAn-source-git, 2022-01-19:
The single-circuit boiler works with an indirect heating boiler for the preparation of hot water (for house). The double-circuit boiler produces hot water immediately (for appartement).

IvAn-source-git, 2022-01-19:
For single-circuit boilers with an indirect heating boiler, the boiler outputs the DHW temperature in ID29.  There may be easier solutions, but I just changed "ID=26: DHW temperature"  and "ID=29: Solar storage temperature"

enver-ablyaev, 2022-02-01:
@IvAn-source-git it didn't help

In addition, I found out that some other data is also absent
[image]

IvAn-source-git, 2022-02-01:
Уже писал. Берите всё с моего репо.  Все котлы разные, со своими датчиками. Мой котел не передает внешнею температуру, давление, температуру обратки, соответственно эти запросы отключены.

rsciriano, 2023-10-21:
Probably fixed by #52

closed 2023-10-21 by rsciriano