rsciriano/ESPHome-OpenTherm #18: Installing in HA

electricfire, 2021-12-24:
I have an issue loading the firmware with the dependency ESP Async WebServer saying fatal error: ESP8266WiFi.h: No such file or directory, removing this dependency loads the firmware fine.  I downloaded the code from Github today 24/12/21

I am using ESP8266 D1 mini.

Am i missing something?

davidrustingha, 2021-12-26:
Same error here.. I'd really like to use this because the simple open therm code from DIYless gives me less options to work with..

Ostoja1970, 2021-12-26:
Same error here..

marcos-acpc, 2021-12-26:
Try check issue nr 5 had the same error commenting the line sorted my issue.

electricfire, 2021-12-26:
i reviewed nr 5 and interestingly i read somewhere else about removing that line and i actually already did that just to confirm in your post and elsewhere online it mentions to comment out / remove  "arduino_version: '2.7.2'".

I am still getting the issue with that line out FYI i am on ESPHome 2021.12.1.

Appreciate your feedback thus far.

marcos-acpc, 2021-12-26:
So try to comment the line:
``platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  ->  #- ESP Async WebServer

electricfire, 2021-12-26:
I did that as i stated in my first comment removing the web server and the arduino version worked, just wanted to know if you had a solution to fix with the webserver?

marcos-acpc, 2021-12-26:
Oh ok I understand better what you mean. I have no solution as it works fine for me like this on home assistant.

electricfire, 2021-12-26:
Fair enough thanks for coming back, just a quick question i have ordered the opentherm module, but just to test i loaded this on the d1 mini and it just loops through connecting and disconnecting i assume this is normal behaviour? 

[14:54:29][D][opentherm_component:095]: update heatingWaterClimate: 0
[14:54:29][D][opentherm_component:096]: update hotWaterClimate: 0
[14:54:32][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)
INFO main-heating.local: Ping timed out!
INFO Disconnected from ESPHome API for main-heating.local
WARNING Disconnected from API
INFO Successfully connected to main-heating.local
[14:54:57][C][template.switch:058]: Template Switch 'PID Climate Autotune'
[14:54:57][C][template.switch:059]:   Restore State: NO
[14:54:57][C][template.switch:060]:   Optimistic: NO
[14:54:57][D][opentherm_component:095]: update heatingWaterClimate: 0
[14:54:57][D][opentherm_component:096]: update hotWaterClimate: 0
[14:55:00][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)
INFO main-heating.local: Ping timed out!
INFO Disconnected from ESPHome API for main-heating.local
WARNING Disconnected from API

marcos-acpc, 2021-12-26:
From what I'm aware you need to have the module connected to the d1 mini and the boiler for.it to work properly, till all was connected it did not work with me

electricfire, 2021-12-26:
great thank you so much for your feedback, good work!

closed 2021-12-26 by electricfire