rsciriano/ESPHome-OpenTherm #21: PID Controller

flodus, 2022-01-04:
Hello,
I've installed your esphome opentherm by commenting arduino version and replace ESP Async WebServer library, by `esphome/ESPAsyncWebServer-esphome @ 2.1.0`
I've got the diyless opentherm shield
and I've got this boiler `https://www.viessmann.fr/fr/chauffage-maison-individuelle/chaudiere-gaz/condensation/vitodens-100-w-vitodens-111-w.html` Vitodens 100 w
I looked a bit at the doc `https://esphome.io/components/climate/pid.html?highlight=climate`

and for the autotune of the PID, they explain `https://esphome.io/components/climate/pid.html?highlight=climate#autotuning`
that we must put the values

```
     control_parameters:
       kp: 0.0
       ki: 0.0
       kd: 0.0

```
and start the autotune
in your opentherm.yaml file there are already the values

```
     control_parameters:
       kp: 0.38197
       ki: 0.01012
       kd: 3.60387

```
should we redo the autotune for us or should we keep your conf ?

thank you in advance for your reply and thank you for your great work

rsciriano, 2022-01-11:
Hi @flodus,

Yes, you should redo the autotune for yourself.

To start the autotune I think you don't need to set the control_parameters to 0 but, when the process is finished, it is important to copy the calculated values to the yaml file and compile and upload the new firmware version.

bartmarchand1, 2022-01-19:
Hello,

There are some issues with the autotuning:
[07:32:50][D][pid.autotune:180]:   Status: Trying to reach nan °C
[07:32:50][D][pid.autotune:181]:   Stats so far:
[07:32:50][D][pid.autotune:182]:     Phases: 0
[07:32:50][D][pid.autotune:183]:     Detected 0 zero-crossings
[07:32:50][D][pid.autotune:185]:     Current Phase Min: nan, Max: nan

the control parameters are set to 0.0

foltymat, 2022-10-23:
Hi @flodus, I have the same boiler and I was wondering - have you managed to run the code successfully with your setup?

flodus, 2022-11-25:
> Hi @flodus, I have the same boiler and I was wondering - have you managed to run the code successfully with your setup?

Hello,
The code is running fine with my configuration... however I failed to configure the pid as I wanted. I don't have an air conditioner and the temperature doesn't come down fast enough for the autotune to give me a satisfactory result.
Especially since I also use a wood stove.
Everything else seems to be working fine.

foltymat, 2022-12-13:
Yeah, I got something similar with my autotune experiment, but the values were so off, that the heating started to act super weird. 
Remember the simple translation for what those values mean.

kp: how big should each change be
ki. how often you want it updated? (above one would be multiple times a second, below 0 is once per minutes)
kd: how far ahead should it think?

I know it's not ideal but at least having the basic understanding helped me start with the fine tuning.
Eventually, I settled on kd: 50 ish

foltymat, 2022-12-13:
My situation is definitely a bit different as I have to switch heating zones thoughout the day. But in your example, yes, if I heat our main living space with everything else closed, it generally takes about an hour to hour and a half to heat it up by one degree. Definitely not the full two hours like in your place.

My current values:
      kp: 0.5
      ki: 0.01
      kd: 30

neforce, 2022-12-13:
Thanks for your insights. I'll keep track of the current config a few (cold days now in Holland), and will change it to KD: 50 (10% of now), at the end of this week, and 'll monitor it again.  Let's see if anything changes (especially to gas usage).