rsciriano/ESPHome-OpenTherm #22: Home Assistant Integration (LoveLace)

electricfire, 2022-01-06:
Hi so i have flashed my esp8266 and all looks ok (FYI i dont have the opentherm connected as i have not received it yet) I can add the device to my integrations but i don't see the additional properties see below 

[image]
[image]

Is it because there is no opentherm connected which i find hard to believe or is there something missing?

rsciriano, 2022-01-06:
Hi @electricfire,

This happens because the opentherm adapter is not connected.

If you want to test without connecting it you can change line 35 of the `opentherm_component.h` file by increasing the polling time. 

```
  OpenthermComponent(): PollingComponent(10000) {
```

electricfire, 2022-01-06:
Awesome thank you so much!!

closed 2022-01-06 by electricfire