rsciriano/ESPHome-OpenTherm #24: Setup

electricfire, 2022-01-09:
Hi got this flashed a week a go dont have the new boiler yet (@rsciriano  thank you so far for solving other questions i have asked) but just getting ahead of the game so that when i do get the boiler (still need to decide which one to get as not all support opentherm and use some kind of interface i am in the UK so if anyone has good experience with a boiler please let me know i was looking at the Viessmann) i have things set up from this end.  There are features like set back temp and min and max water temp which i don't see available, is this something i do via Node-Red or do i need to modify the yaml file to pull in these options, would really appreciate your help on this please.  Also multi-zoning i assume that we would control the boiler on off from Node Red depending which zone is demanding and we just tell the boiler when there is demand required turn on etc as for the PID Auto climate turned on could you explain what this does as i see you mention it to be turned on.

Thank you for your assistance.

marcos-acpc, 2022-01-09:
Hi I'm in the UK too and I have a ideal logic+ and works fine.

electricfire, 2022-01-09:
> Hi I'm in the UK too and I have a ideal logic+ and works fine.

marcos-acpc, 2022-01-23:
it runs as it is but in not expert so I cant say how good it runs. I'm now trying to play with auto-tune but it's not working for me.