rsciriano/ESPHome-OpenTherm #27: No interlock hot water and PID climate controller

ShurikenGitHub, 2022-02-03:
Hi,

When PID climate controller is in "heating" mode, the pump is circulating water in the radiators.
When hot water is required, the three way valve shuts but the pump is still running.
So there is a bang, aka water hammer.
There should be a sort of interlock that turns off the pump before changing the three way valve to hot water.

Hope the problem description is clear to you.

Thanks.

Chupaka, 2022-02-04:
Hello.

Isn't it the boiler itself who should decide when to start/stop its internal components? Or what pump do you mean? Thermostat should not and can not control boiler's pumps/valves directly.

ShurikenGitHub, 2022-02-04:
Hi @Chupaka 

> Isn't it the boiler itself who should decide when to start/stop its internal components?

True, the code cannot control internal components.

> Or what pump do you mean? Thermostat should not and can not control boiler's pumps/valves directly.

I mean the internal pump that circulates the heating water for the radiators.

Maybe if the boiler gets 2 commands at the same time (PID from heating to idle and hot water from off to heating) it doesn't know what to do? Should there maybe be a time delay between commands? I only get the hammer sound if the PID is in "heating" mode, heating water is "off" and hot water turns to "heating". The pump stops almost immediately after the valve is closed.

I never had such water hammer sounds before without the opentherm code so i'm pretty sure it should be the code.

ShurikenGitHub, 2022-02-04:
Hi @rsciriano @Chupaka 
`    else {
      // If the room thermostat is off, set it to 10, so that the pump continues to operate
      heating_target_temperature = 10.0;
      ESP_LOGD("opentherm_component", "setBoilerTemperature at %f °C (default low value)", heating_target_temperature);
    }`

It looks to me like line 129 in opentherm_component is used to keep the pump "running". This sounds to me like it will run the pump after the target temperature is reached to keep water in circulation. This might not be the correct way to do it...? I guess this doesn't turn the pump off before turning the hot water to heat and so causes waterhammer?

rsciriano, 2022-02-05:
> ` else { // If the room thermostat is off, set it to 10, so that the pump continues to operate heating_target_temperature = 10.0; ESP_LOGD("opentherm_component", "setBoilerTemperature at %f °C [...]

Hi @ShurikenGitHub,

The boiler is responsible for controlling the behaviour of its internal component, this only keeps the water in the radiators circulating, even if the room has reached the target temperature.

However, forcing this behaviour was not a good idea, I will remove it in future versions.

eavyon, 2022-02-06:
I'll add I noticed this same behaviour sometimes but didn't really think much of it. Before this I controlled the CV(Vaillant VHR 18-22C) with only ihormelnyk arduino code and I don't think I've heard that banging. 

I have no logging of normal behaviour of a pid controlled thermostat so I don't know if its an accepted way to leave your CV on heating but setting a deliberate low temp(10) when no heat is needed. As soon as the target temperature is lower than the actual temp my CV starts up to heat the hot water. And when central heating is needed immediatly it starts all over again. Which causes a lot of valve switching and igniting the flame in short bursts. Somewhere in between I hear this banging.

I think putting in some delay would also help so the device can figure itself out. I'll try with removing line 129 to test a few days.

eavyon, 2022-02-07:
I've just catch this instant valve change while pump is running. I've added code to see if CH or DHW is enabled.
[image]
DHW just kicked in right af CH because it needs to heat to the internal setpoint to prevent legionella. But because CH is still enabled from the thermostat the pump stays running and cools down the water so the boiler thinks it should go for another round of CH heating. This is just messy and counteracts legionella prevention. 

I've tried myself to change the code so the thermostat while in idle state disables central heating until it is needed again and or setting it to 10 degrees. I wasn't able yet to get this working but this should help. Also having some timeout after CH or DHW demand I think will help.

spattinson, 2022-03-17:
This may be a bit off topic, I am trying to plan my boiler replacement to opentherm and this project as control. 

Three way valves do not have an off position as far as ones i have seen. They have three positions: hot water, heating, and both which is mid position. Valve has two wires that determine position, one for heating and another for hot water. With opentherm it may be better to have two valves instead. One normally open for heating and one normally closed for hot water. Then control them with one wire for both valves, then they are interlocked with one valve open at all times. Then when hot water demand is there boiler fires at max temperature, closes heating valve, opens hot water valve and heats hot water as fast as possible without interfering with heating flow temperature. Then when hot water is heated the opentherm and pid control can properly modulate heating again.

ShurikenGitHub, 2022-03-17:
> I've just catch this instant valve change while pump is running. I've added code to see if CH or DHW is enabled. [image] DHW just kicked in right af CH because it needs to heat to the internal [...]

See my fork. I tried to "optimize" the code to do as little valve switches as possible. It will only change to the hot water position is there is demand for hot water. The main disadvantage is that if you don't use hot water often, the boiler temperature will keep dropping to as low as ~30c for me. This is because the central heating target temp is 10c, so the boiler won't start heating until the internal temp is at 10c. So if you don't use hot water every day, this can cause legionella to grow. Tough you can solve this by turning off and on the domestic hot water thermostat. (maybe do this once every day at midnight with an automation)