rsciriano/ESPHome-OpenTherm #28: Does not work.

roma1141r1, 2022-02-23:
The device itself is visible in the home assistant, but the internal items are not visibl
[image]
e

roma1141r1, 2022-02-23:
When connected to the boiler, the boiler goes into the off state and turns off, i.e. opentherm bus works but esp sends it a command to turn OFF

roma1141r1, 2022-02-23:
This is the code I'm using, but it works the same way by default
`substitutions:
  devicename: opentherm
  upper_devicename: Opentherm

esphome:
  name: baxi-climate
  platform: ESP8266
  board: d1_mini
  platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  includes:
    - esphome-opentherm/
    
    

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.0.27
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1

#captive_portal:
logger:
  #level: ERROR
api:
ota:
#web_server:
  #port: 80
  #auth:
    #username: roma1141
    #password: roma1141

i2c:
  sda: D6
  scl: D7
  scan: true
  id: bus_a

custom_component:
  - lambda: |-
      auto opentherm = new OpenthermComponent();
      return {opentherm};
    
    components:
      - id: opentherm
output:
  - platform: custom
    type: float
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      auto opentherm_pid_output = new OpenthermFloatOutput();
      openthermComp->set_pid_output(opentherm_pid_output);
      App.register_component(opentherm_pid_output);     
      return {opentherm_pid_output};
    outputs:
      id: pid_output
      min_power: 21.00%
      max_power: 75.00%

sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      id: temperature_sensor
     # oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 1s

  - platform: homeassistant
    id: temperature_sensor1
    entity_id: sensor.srednia_temp_v_dome
    name: "srednia_temp_v_dome"
    unit_of_measurement: °C
  - platform: pid
    name: "PID Climate Result"
    type: RESULT    
  - platform: pid
    name: "PID Climate HEAT"
    type: HEAT    
  - platform: pid
    name: "PID Climate ERROR"
    type: ERROR    

  - platform: custom
    lambda: |-    
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      return { 
        openthermComp->boiler_temperature, 
        openthermComp->external_temperature_sensor, 
        openthermComp->return_temperature_sensor, 
        openthermComp->pressure_sensor,
        openthermComp->modulation_sensor,
        openthermComp->heating_target_temperature_sensor
      };
    sensors:
    - name: Boiler Temperature
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "External Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 0  
    - name: "Return Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "Heating Water Pressure"
      unit_of_measurement: hPa
      accuracy_decimals: 1
    - name: "Boiler Modulation"
      unit_of_measurement: "%"
      accuracy_decimals: 0
    - name: "Heating Target Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
      

binary_sensor:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      return {openthermComp->flame};
    binary_sensors:
    - name: "Flame"
      device_class: heat

switch:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      return {openthermComp->thermostatSwitch};
    switches:
      name: "Disable PID"
  - platform: template
    name: "PID Climate Autotune"
    turn_on_action:
      - climate.pid.autotune: pid_climate      

climate:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      return {
        openthermComp->hotWaterClimate, 
        openthermComp->heatingWaterClimate
      };
    climates:
      - id: hot_water
        name: "Hot water"
      - id: heating_water
        name: "Heating water"
  - platform: pid
    id: pid_climate
    name: "PID Climate Controller"
    visual:
      min_temperature: 16 °C
      max_temperature: 28 °C
      temperature_step: 0.01 °C
    sensor: temperature_sensor
    default_target_temperature: 24.1°C
    heat_output: pid_output
    control_parameters:
      kp: 1.0
      ki: 1.007
      kd: 10.01`

roma1141r1, 2022-02-23:
Maybe this log will help, but I doubt it...
`[D][sensor:124]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[D][sensor:124]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Modulation': Sending state -1.00000 % with 0 decimals of accuracy
[D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][climate:385]: 'Hot water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: -1.00°C
[D][climate:414]:   Target Temperature: 0.00°C
[D][climate:385]: 'Heating water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: 0.00°C
[D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[D][api.connection:826]: Home Assistant 2022.2.9 (192.168.0.100): Connected successfully
[D][api.connection:826]: ESPHome Logs 2022.2.4 (192.168.0.100): Connected successfully
[C][logger:233]: Logger:
[C][logger:234]:   Level: DEBUG
[C][logger:235]:   Log Baud Rate: 115200
[C][logger:236]:   Hardware UART: UART0
[D][opentherm_component:104]: update heatingWaterClimate: 0
[D][opentherm_component:105]: update hotWaterClimate: 0
[D][opentherm_component:142]: setBoilerTemperature  at 0.000000 °C (from PID Output)
[D][sensor:124]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[D][sensor:124]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Modulation': Sending state -1.00000 % with 0 decimals of accuracy
[D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][climate:385]: 'Hot water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: -1.00°C
[D][climate:414]:   Target Temperature: 0.00°C
[D][climate:385]: 'Heating water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: 0.00°C
[D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[I][app:102]: ESPHome version 2022.2.4 compiled on Feb 23 2022, 12:35:28
[C][wifi:491]: WiFi:
[C][wifi:353]:   Local MAC: 50:02:91:48:00:C2
[C][wifi:354]:   SSID: [5m'INET'[6m
[C][wifi:355]:   IP Address: 192.168.0.27
[C][wifi:356]:   BSSID: [5mCC:2D:E0:86:C0:57[6m
[C][wifi:358]:   Hostname: 'baxi-climate'
[C][wifi:360]:   Signal strength: -40 dB [0;32m▂▄▆█[0m
[C][wifi:364]:   Channel: 5
[C][wifi:365]:   Subnet: 255.255.255.0
[C][wifi:366]:   Gateway: 192.168.0.1
[D][sensor:124]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[D][opentherm_component:105]: update hotWaterClimate: 0
[D][opentherm_component:142]: setBoilerTemperature  at 0.000000 °C (from PID Output)
[D][sensor:124]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[D][sensor:124]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][sensor:124]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[D][sensor:124]: 'Boiler Modulation': Sending state -1.00000 % with 0 decimals of accuracy
[D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[D][climate:385]: 'Hot water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: -1.00°C
[D][climate:414]:   Target Temperature: 0.00°C
[D][climate:385]: 'Heating water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[... 145 lines cut ...]
[D][climate:385]: 'Hot water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: -1.00°C
[D][climate:414]:   Target Temperature: 0.00°C
[D][climate:385]: 'Heating water' - Sending state:
[D][climate:388]:   Mode: OFF
[D][climate:390]:   Action: OFF
[D][climate:408]:   Current Temperature: 0.00°C
[D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[D][api:102]: Accepted 192.168.0.100
[D][api:102]: Accepted 192.168.0.100
[C][bme280.sensor:143]: BME280:
[C][bme280.sensor:144]:   Address: 0x76
[C][bme280.sensor:156]:   IIR Filter: OFF
[C][bme280.sensor:157]:   Update Interval: 1.0s
[C][bme280.sensor:159]:   Temperature 'BME280 Temperature'
[C][bme280.sensor:159]:     Device Class: 'temperature'
[C][bme280.sensor:159]:     State Class: 'measurement'
[C][bme280.sensor:159]:     Unit of Measurement: '°C'
[C][bme280.sensor:159]:     Accuracy Decimals: 1
[C][bme280.sensor:160]:     Oversampling: 16x
[C][bme280.sensor:161]:     Unit of Measurement: 'hPa'
[D][opentherm_component:105]: update hotWaterClimate: 0
[D][opentherm_component:142]: setBoilerTemperature  at 0.000000 °C (from PID Output)`

Chupaka, 2022-02-23:
What if you remove and re-add the ESP to Home Assistant?

P.S. Yeah, the log is completely unreadable...

roma1141r1, 2022-02-24:
> What if you remove and re-add the ESP to Home Assistant?
> 
> P.S. Yeah, the log is completely unreadable...

Yes, of course, I tried deleting and adding again, but it does not help.
Log now I'll try to copy the log more clearly

roma1141r1, 2022-02-24:
INFO Reading configuration /config/esphome/baxi-climate.yaml...
INFO Starting log output from 192.168.0.27 using esphome API
INFO Successfully connected to 192.168.0.27
[11:39:45][I][app:102]: ESPHome version 2022.2.5 compiled on Feb 24 2022, 10:22:57
[11:39:45][C][wifi:491]: WiFi:
[11:39:45][C][wifi:353]:   Local MAC: 50:02:91:48:00:C2
[11:39:45][C][wifi:354]:   SSID: [redacted]
[11:39:45][C][wifi:355]:   IP Address: 192.168.0.27
[11:39:45][C][wifi:356]:   BSSID: [redacted]
[11:39:45][C][wifi:358]:   Hostname: 'baxi-climate'
[11:39:45][C][wifi:360]:   Signal strength: -52 dB ▂▄▆█
[11:39:45][C][wifi:364]:   Channel: 5
[11:39:45][C][wifi:365]:   Subnet: 255.255.255.0
[11:39:45][C][wifi:366]:   Gateway: 192.168.0.1
[11:39:45][C][wifi:367]:   DNS1: 192.168.0.1
[11:39:45][C][wifi:368]:   DNS2: 0.0.0.0
[11:39:45][D][sensor:124]: 'BME280 Temperature': Sending state 23.82000 °C with 1 decimals of accuracy
[11:39:45][D][sensor:124]: 'PID Climate Result': Sending state 134.05586 % with 1 decimals of accuracy
[11:39:45][D][sensor:124]: 'PID Climate HEAT': Sending state 100.00000 % with 1 decimals of accuracy
[11:39:45][D][sensor:124]: 'PID Climate ERROR': Sending state 28.00007 % with 1 decimals of accuracy
[11:39:45][D][climate:385]: 'PID Climate Controller' - Sending state:
[11:39:45][D][climate:388]:   Mode: HEAT
[11:39:45][D][climate:390]:   Action: HEATING
[11:39:45][D][climate:408]:   Current Temperature: 23.82°C
[11:39:45][D][climate:414]:   Target Temperature: 24.10°C
[11:39:45][D][sensor:124]: 'BME280 Pressure': Sending state 1008.87225 hPa with 1 decimals of accuracy
[11:39:45][D][sensor:124]: 'BME280 Humidity': Sending state 40.17969 % with 1 decimals of accuracy
[11:39:45][D][opentherm_component:104]: update heatingWaterClimate: 0
[11:39:45][D][opentherm_component:105]: update hotWaterClimate: 0
[11:39:46][D][opentherm_component:142]: setBoilerTemperature  at 0.000000 °C (from PID Output)
[11:39:48][D][sensor:124]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[11:39:48][D][sensor:124]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[11:39:48][D][sensor:124]: 'Boiler Temperature': Sending state 44.00000 °C with 1 decimals of accuracy
[11:39:48][D][sensor:124]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[11:39:48][D][sensor:124]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[11:39:48][D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[11:39:48][D][climate:385]: 'Hot water' - Sending state:
[11:39:48][D][climate:388]:   Mode: OFF
[11:39:48][D][climate:390]:   Action: OFF
[11:39:48][D][climate:408]:   Current Temperature: 43.00°C
[11:39:48][D][climate:414]:   Target Temperature: 0.00°C
[11:39:48][D][climate:385]: 'Heating water' - Sending state:
[11:39:48][D][climate:388]:   Mode: OFF
[11:39:48][D][climate:390]:   Action: OFF
[11:39:48][D][climate:408]:   Current Temperature: 44.00°C
[11:39:48][D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[11:39:48][D][api.connection:158]: Home Assistant 2022.3.0b0 (192.168.0.100) requested disconnected
[11:39:48][C][logger:233]: Logger:
[11:39:48][C][logger:234]:   Level: DEBUG
[11:39:48][C][logger:235]:   Log Baud Rate: 115200
[11:39:48][C][logger:236]:   Hardware UART: UART0
[11:39:48][D][opentherm_component:104]: update heatingWaterClimate: 0
[11:39:48][D][opentherm_component:105]: update hotWaterClimate: 0
[11:39:49][D][opentherm_component:142]: setBoilerTemperature  at 0.000000 °C (from PID Output)
[11:39:51][D][sensor:124]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[11:39:51][D][sensor:124]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[11:39:51][D][sensor:124]: 'Boiler Temperature': Sending state 44.00000 °C with 1 decimals of accuracy
[11:39:51][D][sensor:124]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[11:39:51][D][sensor:124]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[11:39:51][D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[11:39:51][D][climate:385]: 'Hot water' - Sending state:
[11:39:51][D][climate:388]:   Mode: OFF
[11:39:51][D][climate:390]:   Action: OFF
[11:39:51][D][climate:408]:   Current Temperature: 43.00°C
[11:39:51][D][climate:414]:   Target Temperature: 0.00°C
[11:39:51][D][climate:385]: 'Heating water' - Sending state:
[11:39:51][D][climate:388]:   Mode: OFF
[11:39:51][D][climate:390]:   Action: OFF
[11:39:51][D][climate:408]:   Current Temperature: 44.00°C
[11:39:51][D][climate:411]:   Target Temperature: Low: 0.00°C High: 0.00°C
[... 177 lines cut ...]
[11:40:11][D][sensor:124]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[11:40:11][D][sensor:124]: 'Heating Target Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[11:40:11][D][climate:385]: 'Hot water' - Sending state:
[11:40:11][D][climate:388]:   Mode: OFF
[11:40:11][D][climate:390]:   Action: OFF
[11:40:11][D][climate:408]:   Current Temperature: 43.00°C
[11:40:11][D][climate:414]:   Target Temperature: 0.00°C
[11:40:11][D][climate:385]: 'Heating water' - Sending state:
[11:40:11][D][climate:388]:   Mode: OFF
[11:40:11][D][climate:390]:   Action: OFF
[11:40:11][D][climate:408]:   Current Temperature: 44.00°C
[... cut ...]
[11:40:11][D][sensor:124]: 'PID Climate Result': Sending state 119.37988 % with 1 decimals of accuracy
[11:40:11][D][sensor:124]: 'PID Climate HEAT': Sending state 100.00000 % with 1 decimals of accuracy
[11:40:11][D][sensor:124]: 'PID Climate ERROR': Sending state 27.00005 % with 1 decimals of accuracy
[11:40:11][D][climate:385]: 'PID Climate Controller' - Sending state:
[11:40:11][D][climate:388]:   Mode: HEAT
[11:40:11][D][climate:390]:   Action: HEATING
[11:40:11][D][climate:408]:   Current Temperature: 23.83°C
[11:40:11][D][climate:414]:   Target Temperature: 24.10°C``

roma1141r1, 2022-02-24:
In this case, the boiler is immediately in the off state. And I can't turn it on until I plug in another thermostat.
Although if I connect any other OT thermostat, the boiler will continue to work. Then I connect esp home and the boiler immediately goes into the OFF state
those. The boiler is given an OFF command (because if you simply disconnect the OT thermostat from the boiler, it continues to work for about a minute without a thermostat at all)

roma1141r1, 2022-02-25:
I think the whole problem is that there is no interface from the ESPHOME board inside the Homeassistant, and I cannot set the regulator, therefore the boiler is switched to the OFF state, but I cannot understand why there is no regulator interface in the homeassistant ..

rsciriano, 2022-02-25:
> I think the whole problem is that there is no interface from the ESPHOME board inside the Homeassistant, and I cannot set the regulator, therefore the boiler is switched to the OFF state, but I cannot understand why there is no regulator interface in the homeassistant ..

Hi @roma1141r1, I thing the same. 

In the logs, we can see how Home Assistant is disconected

> [11:39:48][D][api.connection:158]: Home Assistant 2022.3.0b0 (192.168.0.100) requested disconnected

and also, how the comunication with the boiler works properly

> [11:40:11][D][sensor:124]: 'Boiler Temperature': Sending state 44.00000 °C with 1 decimals of accuracy

You can try to update the esphome and Homeassistant versiona to the last stable versions.

roma1141r1, 2022-02-25:
Thank you for your time, but I have updated the ESPHome version
Current version: 2022.2.5
and updated the ESPHOME board
it didn't work

roma1141r1, 2022-02-25:
> > I think the whole problem is that there is no interface from the ESPHOME board inside the Homeassistant, and I cannot set the regulator, therefore the boiler is switched to the OFF state, but I cannot understand why there is no regulator interface in the homeassistant ..
> 
> Hi @roma1141r1, I thing the same.
> 
> In the logs, we can see how Home Assistant is disconnected
> 
> > [11:39:48][D][api.connection:158]: Home Assistant 2022.3.0b0 (192.168.0.100) requested disconnected
> 
> and also, how the communication with the boiler works properly
> 
> > [11:40:11][D][sensor:124]: 'Boiler Temperature': Sending state 44.00000 °C with 1 decimals of accuracy
> 
> You can try to update the esphome and Homeassistant versiona to the last stable versions.

Thank you for your time, but I have updated the ESPHome version
Current version: 2022.2.5
and updated the ESPHOME board
it didn't work

roma1141r1, 2022-02-25:
Sorry for my impoliteness....
Hello!

roma1141r1, 2022-02-25:
Log homeassistant:

Logger: homeassistant.components.esphome
Source: components/esphome/__init__.py:290
Integration: ESPHome (documentation, issues)
First occurred: 15:21:08 (435 occurrences)
Last logged: 19:09:46

Error getting initial data for 192.168.0.27: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
Error getting initial data for 192.168.0.27: Timeout waiting for response for <class 'api_pb2.DeviceInfoRequest'>

rsciriano, 2022-02-25:
Hi @roma1141r1,

I think the problem might be that your boiler does not support the opentherm commands for External Temperature, Return Temperature and Heating Water Pressure

You can try to increase the pooling time by changing line 35 in `opentherm_component.h` file

```c
// Set 3 sec. to give time to read all sensors (and not appear in HA as not available)
 OpenthermComponent(): PollingComponent(3000) {
```

Also, you can remove the code that reads the values of these sensors.

roma1141r1, 2022-02-26:
> Hi @roma1141r1,
>
> I think the problem might be that your boiler does not support the opentherm commands for External Temperature, Return Temperature and Heating Water Pressure
>
> You can try to [...]

Hey! Thanks I will try this.

roma1141r1, 2022-02-26:
Thanks, removing unnecessary sensors and increasing the time by 7000 ms helped

closed 2022-02-26 by roma1141r1