rsciriano/ESPHome-OpenTherm #29: Setting upper an lower limits in boiler

bigbrother72, 2022-02-28:
Hello

My boiler have limits 45-85C.  But i want to set in opentherm PID settings limits 50-75C for water in boiler.

How i can set upper and lower limits in opentherm temostat?

rsciriano, 2022-03-01:
Hi @bigbrother72 ,

You can do it like this:

```yaml
climate:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) opentherm;
      return {
        openthermComp->hotWaterClimate, 
        openthermComp->heatingWaterClimate
      };
    climates:
      - id: hot_water
        name: "Hot water"
      - id: heating_water
        name: "Heating water"
        visual:
          min_temperature: 30 °C
          max_temperature: 85 °C        
```

The custom climates support the same configuration as [core climate](https://esphome.io/components/climate/index.html#config-climate)

bigbrother72, 2022-03-02:
It is only visual on frontend page in HA as i think

`visual (Optional): Visual settings for the climate device - these do not affect operation and are solely for controlling how the climate device shows up in the frontend.

    min_temperature (Optional, float): The minimum temperature the climate device can reach. Used to set the range of the frontend gauge.

    max_temperature (Optional, float): The maximum temperature the climate device can reach. Used to set the range of the frontend gauge.`

rsciriano, 2022-03-02:
The boiler temperature is controlled via `Heating water climate˜  

With this climate you set the low and high values (for example 23 °C and 65 °C)

[image]

- When the room termerature is very low than target temperature, the output of the PID will be 1, and the opentherm component set the boiler terperature to 65 °C 
- When the room termerature is near target temperature, the output of the PID will be near of 0, and the opentherm component set  the boiler terperature to 23 °C 
- When the room termerature is equal or upper target temperature, the output of the PID will be 0, and the opentherm component set the boiler terperature to 10 °C _(this is to keep the pump running, but it was not a good idea and I will change it in future versions)_

bigbrother72, 2022-03-04:
> * When the room termerature is very low than target temperature, the output of the PID will be 1, and the opentherm component set the boiler terperature to 65 °C
>
>     * When the room termerature is [...]

but as I understand from this description, it looks more like a simple thermostat.

First, when we need to bring it to the target temperature, the boiler operates at full power (75 degrees as in screenshot)

When the target temperature is almost reached, the PID output is near zero, and then the boiler simply keeps the heating water temperature at a minimum, but it is not enough to maintain the target temperature in the room.

After some time, when the room temperature drops below the target temperature, the boiler will turn on again at full power.

Or is it because the pid coefficients are wrong?

In the screenshot, the temperature of the water in the boiler and boiler modlation. 
(there was boiler used ащк shower between 21:20 and 21:30.)

[image]

[image]

rsciriano, 2022-03-04:
Hi @bigbrother72 ,

You have to adjust the PID coefficients, using the auto tune feature od PID.

For better monitoring of PID operation, you can use "Heating Target Temperature" or "PID Climate HEAT"

P.D: I have not yet been able to adjust the PID to work fine in my home :(