rsciriano/ESPHome-OpenTherm #32: Remeha Tzerra Ace - Current Temperature: NaN

jfranken2, 2022-07-01:
Hello,

first i want to thank you for this great project. Realy good work!
I'm trying to use the opentherm master shield with a wemos d1 mini from diyless.com with a remeha tzerra ace boiler.

The hot water control and nearly all sensor values in HA are working fine, except the heating target temperature is always set to 10°C and can't be modified via home-assistant. For me it seems to be an issue because the PID_OUTPUT is always 0.0. 

Here is the log of the wemos d1:
```
[14:36:24][C][logger:275]: Logger:
[14:36:24][C][logger:276]:   Level: DEBUG
[14:36:24][C][logger:277]:   Log Baud Rate: 115200
[14:36:24][C][logger:278]:   Hardware UART: UART0
[14:36:24][C][template.sensor:023]: Template Sensor 'Actual temperature interval'
[14:36:24][C][template.sensor:023]:   State Class: ''
[14:36:24][C][template.sensor:023]:   Unit of Measurement: ''
[14:36:24][C][template.sensor:023]:   Accuracy Decimals: 1
[14:36:24][C][template.sensor:024]:   Update Interval: 1.0s
[14:36:24][C][template.switch:058]: Template Switch 'PID Climate Autotune'
[14:36:24][C][template.switch:059]:   Restore State: NO
[14:36:24][C][template.switch:060]:   Optimistic: NO
[14:36:24][C][pid.climate:062]: PID Climate 'PID Climate Controller'
[14:36:24][C][pid.climate:063]:   Control Parameters:
[14:36:24][C][pid.climate:064]:     kp: 0.38197, ki: 0.01012, kd: 3.60387
[14:36:24][C][pid.sensor:056]: PID Climate Sensor 'PID Climate Result'
[14:36:24][C][pid.sensor:056]:   State Class: 'measurement'
[14:36:24][C][pid.sensor:056]:   Unit of Measurement: '%'
[14:36:24][C][pid.sensor:056]:   Accuracy Decimals: 1
[14:36:24][C][pid.sensor:056]:   Icon: 'mdi:gauge'
[14:36:24][C][pid.sensor:056]: PID Climate Sensor 'PID Climate HEAT'
[14:36:24][C][pid.sensor:056]:   State Class: 'measurement'
[14:36:24][C][pid.sensor:056]:   Unit of Measurement: '%'
[14:36:24][C][pid.sensor:056]:   Accuracy Decimals: 1
[14:36:24][C][pid.sensor:056]:   Icon: 'mdi:gauge'
[14:36:24][C][pid.sensor:056]: PID Climate Sensor 'PID Climate ERROR'
[14:36:24][C][pid.sensor:056]:   State Class: 'measurement'
[14:36:24][C][pid.sensor:056]:   Unit of Measurement: '%'
[14:36:24][C][pid.sensor:056]:   Accuracy Decimals: 1
[14:36:24][C][pid.sensor:056]:   Icon: 'mdi:gauge'
[14:36:24][C][captive_portal:088]: Captive Portal:
[14:36:24][C][mdns:084]: mDNS:
[14:36:24][C][mdns:085]:   Hostname: heizungssteuerung
[14:36:24][C][ota:085]: Over-The-Air Updates:
[14:36:24][C][ota:086]:   Address: heizungssteuerung.local:8266
[14:36:24][C][api:138]: API Server:
[14:36:24][C][api:139]:   Address: heizungssteuerung.local:6053
[14:36:24][C][api:143]:   Using noise encryption: NO
[14:36:24][C][homeassistant.sensor:030]: Homeassistant Sensor 'Actual temperature'
[14:36:24][C][homeassistant.sensor:030]:   State Class: ''
[14:36:24][C][homeassistant.sensor:030]:   Unit of Measurement: '°C'
[14:36:24][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[14:36:24][C][homeassistant.sensor:031]:   Entity ID: 'sensor.temperature_sensor'
[14:36:24][D][sensor:124]: 'Actual temperature interval': Sending state nan  with 1 decimals of accuracy
[14:36:24][D][sensor:124]: 'PID Climate Result': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:24][D][sensor:124]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:24][D][sensor:124]: 'PID Climate ERROR': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:24][D][climate:385]: 'PID Climate Controller' - Sending state:
[14:36:24][D][climate:388]:   Mode: HEAT
[14:36:24][D][climate:390]:   Action: IDLE
[14:36:24][D][climate:408]:   Current Temperature: nan°C
[14:36:24][D][climate:414]:   Target Temperature: 25.00°C
[14:36:25][D][opentherm_component:095]: update heatingWaterClimate: 3
[14:36:25][D][opentherm_component:096]: update hotWaterClimate: 3
[14:36:25][D][opentherm_component:125]: setBoilerTemperature  at 10.000000 °C (from PID Output)
[14:36:26][D][sensor:124]: 'External Temperature': Sending state 18.57812 °C with 0 decimals of accuracy
[14:36:26][D][sensor:124]: 'Return Temperature': Sending state 21.96875 °C with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'Boiler Temperature': Sending state 22.08984 °C with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'Heating Water Pressure': Sending state 1.39844 hPa with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[14:36:26][D][sensor:124]: 'Heating Target Temperature': Sending state 10.00000 °C with 1 decimals of accuracy
[14:36:26][D][climate:385]: 'Hot water' - Sending state:
[14:36:26][D][climate:388]:   Mode: HEAT
[14:36:26][D][climate:390]:   Action: OFF
[14:36:26][D][climate:408]:   Current Temperature: 50.32°C
[14:36:26][D][climate:414]:   Target Temperature: 50.00°C
[14:36:26][D][climate:385]: 'Heating water' - Sending state:
[14:36:26][D][climate:388]:   Mode: HEAT
[14:36:26][D][climate:390]:   Action: OFF
[14:36:26][D][climate:408]:   Current Temperature: 22.09°C
[14:36:26][D][climate:411]:   Target Temperature: Low: 5.00°C High: 80.00°C
[14:36:26][D][sensor:124]: 'Actual temperature interval': Sending state nan  with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'PID Climate Result': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:26][D][sensor:124]: 'PID Climate ERROR': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:26][D][climate:385]: 'PID Climate Controller' - Sending state:
[14:36:26][D][climate:388]:   Mode: HEAT
[14:36:26][D][climate:390]:   Action: IDLE
[14:36:26][D][climate:408]:   Current Temperature: nan°C
[14:36:27][D][climate:414]:   Target Temperature: 25.00°C
[14:36:27][D][api:102]: Accepted 192.168.178.5
[14:36:27][D][api.connection:861]: Home Assistant 2022.6.7 (192.168.178.5): Connected successfully
[14:36:27][D][sensor:124]: 'Actual temperature interval': Sending state nan  with 1 decimals of accuracy
[14:36:27][D][sensor:124]: 'PID Climate Result': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:27][D][sensor:124]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:27][D][sensor:124]: 'PID Climate ERROR': Sending state 0.00000 % with 1 decimals of accuracy
[14:36:27][D][climate:385]: 'PID Climate Controller' - Sending state:
[14:36:27][D][climate:388]:   Mode: HEAT
[14:36:27][D][climate:390]:   Action: IDLE
[14:36:27][D][climate:408]:   Current Temperature: nan°C
[14:36:27][D][climate:414]:   Target Temperature: 25.00°C
```

Is there something i forgot to configure, or to modify in the code, or any other ideas?
Thank you.
Regards 
Jan

ShurikenGitHub, 2022-07-19:
Hi Jan,
If the heating is off (seems like your room temperature is 21.7c), the target temp is 10c, if you turn on the heating and set the target temp to 24c or so, the target temp should go to 50c. The target temp is 10c when no heat is required, so in this case it's correct.
If the temperature is still not set, you can have a look at this table, see if all required parameters are supported for your boiler:
https://otgw.tclcode.com/matrix.cgi#boilers

jfranken2, 2022-08-04:
Yes, maybe that is the problem, but i don't know how to set the room temperature. Is it possible to define the room temperature via the esp opentherm shield? I want to replaced my eTwist-thermostat with the esp opentherm shield for controlling everything via HA, but i have only climate.hot_water, climate.heating_water and the pid.pid_climate_controller controls in HA for setting temperatures.

The controls in HA are set currently to the following values:
[image]

The log shows continously the following:
```
[15:52:13][D][sensor:124]: 'External Temperature': Sending state 29.44922 °C with 0 decimals of accuracy
[15:52:13][D][sensor:124]: 'Return Temperature': Sending state 25.51953 °C with 1 decimals of accuracy
[15:52:13][D][sensor:124]: 'Boiler Temperature': Sending state 25.91797 °C with 1 decimals of accuracy
[15:52:13][D][sensor:124]: 'Heating Water Pressure': Sending state 1.29688 hPa with 1 decimals of accuracy
[15:52:13][D][sensor:124]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[15:52:13][D][sensor:124]: 'Heating Target Temperature': Sending state 10.00000 °C with 1 decimals of accuracy
[15:52:13][D][climate:385]: 'Hot water' - Sending state:
[15:52:13][D][climate:388]:   Mode: HEAT
[15:52:13][D][climate:390]:   Action: OFF
[15:52:13][D][climate:408]:   Current Temperature: 47.59°C
[15:52:13][D][climate:414]:   Target Temperature: 50.00°C
[15:52:13][D][climate:385]: 'Heating water' - Sending state:
[15:52:13][D][climate:388]:   Mode: HEAT
[15:52:13][D][climate:390]:   Action: OFF
[15:52:13][D][climate:408]:   Current Temperature: 25.92°C
[15:52:13][D][climate:411]:   Target Temperature: Low: 30.00°C High: 40.00°C
[15:52:14][D][opentherm_component:095]: update heatingWaterClimate: 3
[15:52:14][D][opentherm_component:096]: update hotWaterClimate: 3
[15:52:15][D][opentherm_component:123]: setBoilerTemperature  at 10.000000 °C (from PID Output)
```
Thanks.

ShurikenGitHub, 2022-08-04:
Aha, in line 82 of opentherm.yaml, you need to replace sensor.temperature_sensor with the temperature sensor from home assistant that you want to use as the temperature sensor. So in my case I have an esp temperature sensor in my livingroom with the name sensor.livingroom_temperature_sensor and I replaced that in line 82. If it works you should see the temperature in the  middle of the PID climate controller thermostat card.

jfranken2, 2022-08-04:
Perfect, that was the solution. 
Many thanks! Now it is working 👍

closed 2022-08-04 by jfranken2