rsciriano/ESPHome-OpenTherm #36: external temperature

daninvencions, 2022-10-14:
Hello, sorry for my English. my question is if it is possible to send the outside temperature data from a sensor that is inside homeassistant to the boiler. From what I see, you can read the temperature of the sensor connected to the boiler, but if you don't have one connected, can you send the temperature of another sensor? I understand that to improve the operation of the cladera, the boiler should have this information in order to modulate the flame accordingly. Thank you very much in advance

Chupaka, 2022-11-22:
According to OpenTherm Protocol Specification v2.2, outside temperature is read-only. You cannot write this parameter.

daninvencions, 2022-11-22:
it's okay. Thank you very much for the reply.