rsciriano/ESPHome-OpenTherm #37: Lack of documentation

foltymat, 2022-10-22:
Hi, let me start by saying THANKS for your work! I was looking for an ESPhome based system to use with OpenTherm and this seems to be the one.

Honestly, I'm not sure how often you read these, so my hopes are not that high :)

I need to address this issue with the lack of documentation of this whole program. I'd consider myself an advanced beginner when it comes to code, but figuring out how this works was a chore and a half.

My main issue is that I still haven't figures out how to properly control the boiler after everything has been connected.
In my case, my Viessmann boiler does not support 'Returning temperarure' nor 'Heating water pressure.' But I imagnine that some parameters are crucial for the thing to work and some are not...

### The three climate controls

So how exactly am I supposed to know the difference between 'Heating water' 'How water' and 'PID Climate control' ?
Through testing I did figure the most basic things out, but not really with a lot of details.

In the end, my hope would be that after setting the PID Climate Control, stuff would begin to happen. But no, I had to figure out that you can change the entity ID of the actual temperature to a one from Home Assistant - but even then, nothing happened.

### Disable PID and Autotune

No idea what disable PID does (from my testing - nothing)
And I had to learn about the Autotune from ESPhome documentation - which, fair enough, but there's no reference to it anywhere.

### Summary

In the end, I'm just frustrated. I WANT this to work so badly but with no real documentation I don't know how.
And with my limited knowledge in coding it's virtually impossible to find this out on my own.

foltymat, 2022-10-24:
Right, as long as nobody responds, I'll keep this as an archive of what I found during testing, how all of this works, and share this with everyone who might be struggling like I did (and do :) )

All of these notes are based on [v0.4](https://github.com/rsciriano/ESPHome-OpenTherm/releases/tag/v0.4) released on **January 6th 2022**. It seems that @rsciriano IS still working on this as there are a lot of builds being made in the meantime; so there may be some light at the end of the tunnel and some more functionality coming.

As I am the n00b in all of this let me start from the beginning.

**The BASICS**

This sketch allows you to have a bridge between ESPhome (including Home Assistant) and your OpenTherm enabled boiler.
In my case, that is a Viessmann Vitodens 100-w b1hc-35, but many others support it.

If you're just starting out, start by looking at the [compatibility matrix here](https://otgw.tclcode.com/matrix.cgi#boilers). What you will be looking at is a page made by the creator of the original OpenTherm Gateway. If I understood correctly, this is the backbone of any OpenTherm implementation we see today - even in this ESPhome version.

Provided you are using Home Assistant like I am, there is a [great alternative approach on the DIYLESS website](https://diyless.com/blog/home-assistant-opentherm-thermostat), if you ultimately fail to establish this code. The difference is, the underlying code and the method of communication. Using this method, you will be using Arduino code and MQTT to communicate with either Home Assistant, or a system of your choice (I haven't tried this code at all, though. At least not yet.)

The ESPversion like this one, however, is using the native ESPhome API to communicate with Home Assistant and I am much more comfortable with YAML, so that's why I decided to try this first.

**What I bought**

My biggest worry was the definition between the three options presented to me.
That would be: 1) OpenTherm Gateway, 2) OpenTherm Master Shield, 3) OpenTherm Slave Shield.

Honestly, I'm still not sure I got this 100% correct so take it with a grain of salt.
(All links are to the DIYLESS website where I got my Master shield)

You use [master shield](https://diyless.com/product/master-opentherm-shield) in order to have a full control over your boiler. You get access to all of these fun parameters of your heater and you can fiddle with almost anything - that's why I went this way. 
_Remember, your boiler has a mind of its own, so there is **almost zero** chance you would break something._

The [Slave shield](https://diyless.com/product/slave-opentherm-shield) would be used to connect to your existing OpenTherm thermostat. Honestly, I still haven't figured ouut how would you use this on its own.

The [OpenTherm Gateway](https://diyless.com/product/esp8266-opentherm-gateway) combines the Master and the Slave into a single device. Meaning you would connect to your boiler AND to your existing Thermostat.

After reading all of this, my original thought was "I need to get that Gateway." But since my use case will be using smart Zigbee TRVs on my Radiators, the Thermostat portion will be done completely inside Home Assistant.

**The INSTALLATION**

I cannot say much about the installation part as that is extermely simple! As the Readme states, you just copy the repository into your esphome folder and you're off to the races. Just make sure you open the opentherm.yaml file and change the devicename (line 2) to your liking.

Now, my personal issue was that instead of using a Wemos D1 mini (like recommended), I used a generic NodeMCU board. It works, but make sure you change the board type on line 8 to nodemcuv2

**Here's the original:**
```
substitutions:
  devicename: opentherm
  upper_devicename: Opentherm

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  includes:
    - esphome-opentherm/
```

**Here's mine:** (Yes, for some reason, it wouldn't compile with the board indented inside esphome so I moved it)
```
substitutions:
  devicename: 0p-kotelna-opentherm
  upper_devicename: Opentherm

esphome:
  name: $devicename
  platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  includes:
    - esphome-opentherm/
[... 47 lines cut ...]
I urge you to read a proper how-to on the ESPhome website, but here's the jist.

First, make sure your system is doing something! Only then, set a temperature on the PID Controller to something you can reach and then turn on the 'PID Climate Autotune' switch that you will see in there as well.

To see progress of the autotune, you will need to see the logs of the ESPhome device. You will be looking at something like this:
```
   PID Autotune:
     Autotune is still running!
     Status: Trying to reach 24.25 °C
     Stats so far:
       Phases: 4
       Detected 5 zero-crossings
       # ...
```
For example, in the output above, the autotuner is driving the heating output at 100%
and trying to reach 24.25 °C.

This will continue for some time until data for 6 phases (or a bit more, depending on the data
quality) have been acquired. 

For me, in a big house and old heavy radiators, this took almost 24 hours! So just be patient.
In the end, it will spit out values based on the algorith you should copy to your yaml file at the very end, recompile and upload.

### Summary
And that's it. This has been a chore to write down but I hope someone in the future will find this and appreciate at least the bit of work I did to find this out! 👍

rsciriano, 2022-10-29:
Hi @foltymat, 

Thank you very much for your contribution, I added a link to this issue in the readme. 

I am working on a [new completely rewritten version](https://github.com/rsciriano/ESPHome-OpenTherm/tree/external-component), which I will try to document in detail, your work will be very useful for that :-)

foltymat, 2022-10-31:
No worries, I was just frustrated myself that I couldn't get it to work from the beginning so at least I helped some people later on. 

I will be watching your work on the other version! If you want, I can help with the documentation once you're at a comfortable state. Reach out, we can always do something :)

spasayrim, 2023-02-19:
> To see progress of the autotune, you will need to see the logs of the ESPhome device. You will be looking at something like this:
>
> ```
>    PID Autotune:
>      Autotune is still running!
>      Status: [...]

[W][pid.climate:167]: pid_climate_controller: !!! For PID autotuner you need to set AUTO (also called heat/cool) mode!
Any ideas to fix this problem?

craigsbits, 2023-02-25:
@foltymat Thankyou for putting what I feel in words.  Something I cannot get to grips with 'yet' is the PID part.  I understand all about PID controllers and auto-tune, what I don't get is why a modern Opentherm boiler would need this.  The boiler has the PID controller and modulates the central heating temperature to achieve a control of the CH water which does not keep switching on and off - like a simple mechanical thermostat does.  I want the Thermostat (Master) (the boiler is the slave in this case) to tell the boiler what temperature I want and what the current temperature from whatever sensor we might have (I use the square Xiaomi Bluetooth sensors shown on the Diyless website - suitably reprogrammed for ESPhome! (see Diyless website.  Its the easiest hack I have ever done!).  I cannot understand why the boiler cannot do the PIDing.  However....Opentherm is not just for boilers and I am sure some 'heaters'/'coolers' ovens who knows will actually benefit (be more efficient) from a PID controller maybe thats why its there?

straaat, 2023-02-26:
The boiler is not a room thermostat. It might use PID-control to determine how much gas to burn to reach the CH water temperature the thermostat asked for. 

There need to be some logic to convert a (or multiple) room temperature input to a CH water temperature output. This is traditionally the room thermostats job.

craigsbits, 2023-02-26:
Traditionally the Thermostats did not communicate temperature just demand.  The boiler now (opentherm) gets the actual temperature (or an average of several thermometers) it does the modulating (pid) to achieve a stable temperature.  If that temperature is jumping about because another pid is trying to be clever it is going to struggle.

straaat, 2023-02-26:
My boiler does not get the room temperature. 
It gets a requested CH water temperature. It modulates gas burning to achieve the requested water temperature. 
It is the job of the room thermostat to decide what water temperature to request. 

There might be other systems where the boiler can just receive a room air temperature through OpenTherm, I don’t know about that. In that case you would just need a device to relay sensor data, not a thermostat. I guess this software is not for you in that case.

craigsbits, 2023-02-26:
Ok.  I am wrong! Thanks Straaat for correcting me.  Every day is a learning experience.  I went back and watched the youtube videos again!!  The pid is calculating a new flow temperature to tell the 'dumb' boiler.  So there is only one pid and no 'struggle' as I incorrectly said.  I thought it was as you said in second paragraph but having looked at my boiler (Intergas xclusive) it has no idea what the actual room temperature is, my Worcester Bosch did have a room temperature sensor and modulated something from this input.

aistodorescu, 2023-02-27:
I think the boiler still does a PID to keep the water at the desired setpoint (it does this modifying the boiler flame modulation). It would be very nice if we could use the ESPHome percentage to set directly the boiler modulation.
I have tried "Maximum relative modulation level setting"(ID=14) on my Daikin D2CND but it doesn't have any impact.

Chupaka, 2023-02-27:
> I have tried "Maximum relative modulation level setting"(ID=14) on my Daikin D2CND but it doesn't have any impact.

The same for my Bosch Gaz 6000 - boiler accepts it but does nothing, actual modulation level goes higher than that maximum value.

> It would be very nice if we could use the ESPHome percentage to set directly the boiler modulation.

Not via OpenTherm. They don't allow thermostats to control internal behaviour of a boiler. So if your boiler blows up because of an overload (like 100% modulation over long time), that's a failure of the boiler, not any 3rd party device connected to it :)

craigsbits, 2023-02-28:
Not at all sure why you would want to modulate max boiler modulation but this but probably is down to the limited Opentherm writeable data objects, but maybe for you it can be indirectly controlled or at least improved.  Firstly though your boilers should have a maximum pump speed (head or %full flow) which is set during commissioning by the installer on installers menu.  Most boilers are just too powerful on Central Heating.  A typical house (in uk) should be set to overcome a 7kw 'heat loss' to maintain say 22 deg. inside with 2 deg. outside and overcome pipe runs size rads and internal resistance.  I believe this setting will not be on Opentherm but get this right first! will cause cycling problems if it is too high.  In my case for the 36kw Intergas its about 30% of max. Pump speed.  This should stop the boiler switching on/ off significantly this is what is getting modulated.

spasayrim, 2023-02-28:
[16:25:09][D][climate:387]: 'PID Climate Controller' - Sending state:
[16:25:09][D][climate:390]:   Mode: HEAT
[16:25:09][D][climate:392]:   Action: HEATING
[16:25:09][D][climate:410]:   Current Temperature: 23.69°C
[16:25:09][D][climate:416]:   Target Temperature: 23.50°C
[16:25:09][D][pid.autotune:185]: pid_climate_controller: PID Autotune:
[16:25:09][D][pid.autotune:186]:   Autotune is still running!
[16:25:09][D][pid.autotune:187]:   Status: Trying to reach 23.75 °C
[16:25:09][D][pid.autotune:188]:   Stats so far:
[16:25:09][D][pid.autotune:189]:     Phases: 1
[16:25:09][D][pid.autotune:190]:     Detected 1 zero-crossings
[16:25:09][D][pid.autotune:192]:     Current Phase Min: -0.20, Max: 0.27
[16:25:09][D][sensor:127]: 'Actual temperature interval': Sending state 23.69000  with 1 decimals of accuracy
[16:25:09][D][sensor:127]: 'PID Climate Result': Sending state 100.00000 % with 1 decimals of accuracy
[16:25:09][D][sensor:127]: 'PID Climate HEAT': Sending state 100.00000 % with 1 decimals of accuracy
[16:25:09][D][sensor:127]: 'PID Climate ERROR': Sending state -19.00005 % with 1 decimals of accuracy
[16:25:10][D][opentherm_component:095]: update heatingWaterClimate: 3
[16:25:10][D][opentherm_component:096]: update hotWaterClimate: 3
[16:25:11][D][opentherm_component:123]: setBoilerTemperature  at 66.000000 °C (from PID Output)
[16:25:12][D][sensor:127]: 'External Temperature': Sending state -1.00000 °C with 0 decimals of accuracy
[16:25:12][D][sensor:127]: 'Return Temperature': Sending state -1.00000 °C with 1 decimals of accuracy
[16:25:12][D][sensor:127]: 'Boiler Temperature': Sending state 67.09766 °C with 1 decimals of accuracy
[16:25:12][D][sensor:127]: 'Heating Water Pressure': Sending state -1.00000 hPa with 1 decimals of accuracy
[16:25:12][D][sensor:127]: 'Boiler Modulation': Sending state 0.00000 % with 0 decimals of accuracy
[16:25:12][D][sensor:127]: 'Heating Target Temperature': Sending state 66.00000 °C with 1 decimals of accuracy
[16:25:12][D][climate:387]: 'Hot water' - Sending state:
[16:25:12][D][climate:390]:   Mode: HEAT
[16:25:12][D][climate:392]:   Action: OFF
[16:25:12][D][climate:410]:   Current Temperature: 33.40°C
[16:25:12][D][climate:416]:   Target Temperature: 51.00°C
[16:25:12][D][climate:387]: 'Heating water' - Sending state:
[16:25:12][D][climate:390]:   Mode: HEAT
[16:25:12][D][climate:392]:   Action: HEATING
[16:25:12][D][climate:410]:   Current Temperature: 67.10°C
[16:25:12][D][climate:414]:   Target Temperature: Low: 50.00°C High: 66.00°C

The log says: "Mode: HEAT"
Where can I edit this setting to "AUTO" so I can run "PID Autotune" to get data: kp, ki, kd

Now when I run "PID Autotune" the log says this:
[W][pid.climate:167]: pid_climate_controller: !!! For PID autotuner you need to set AUTO (also called heat/cool) mode!