rsciriano/ESPHome-OpenTherm #48: noob starting out: nothing shows up

unlucio, 2023-02-03:
Hello,
today I received the [DIYLESS](https://diyless.com/product/esp8266-thermostat-shield) opentherm shield, attached it to my D1 mini and started experimenting.
I tried both the version in the `master` branch as well as the experimental `external-component` one, but nothing happens 😶
HA found a new ESP device to add, tho once I add it and go to the device page this is what I get:
[image]
I wasn't expecting it to work on the 1st try, but I was expecting to see at least some sensors with no-value or weird values.
I really don't know where to go form here so if some one can kindly point me in the right direction it would be awesome.

I went and check the logs, and here's what I get:
```
[19:25:09][I][app:102]: ESPHome version 2022.12.8 compiled on Feb  3 2023, 18:12:31
[19:25:09][C][wifi:504]: WiFi:
[19:25:09][C][wifi:362]:   Local MAC: 8C:AA:B5:CF:A6:E7
[19:25:09][C][wifi:363]:   SSID: [redacted]
[19:25:09][C][wifi:364]:   IP Address: 10.6.9.95
[19:25:09][C][wifi:365]:   BSSID: [redacted]
[19:25:09][C][wifi:367]:   Hostname: 'boiler'
[19:25:09][C][wifi:369]:   Signal strength: -47 dB ▂▄▆█
[19:25:09][C][wifi:373]:   Channel: 9
[19:25:09][C][wifi:374]:   Subnet: 255.255.255.0
[19:25:09][C][wifi:375]:   Gateway: 10.6.9.1
[19:25:09][C][wifi:376]:   DNS1: 10.6.9.2
[19:25:09][C][wifi:377]:   DNS2: 0.0.0.0
[19:25:09][D][sensor:126]: 'Actual temperature interval': Sending state nan  with 1 decimals of accuracy
[19:25:09][D][sensor:126]: 'PID Climate Result': Sending state 0.00000 % with 1 decimals of accuracy
[19:25:10][D][sensor:126]: 'PID Climate HEAT': Sending state 0.00000 % with 1 decimals of accuracy
[19:25:10][D][sensor:126]: 'PID Climate ERROR': Sending state 0.00000 % with 1 decimals of accuracy
[19:25:10][D][climate:385]: 'PID Climate Controller' - Sending state:
[19:25:10][D][climate:388]:   Mode: HEAT
[19:25:10][D][climate:390]:   Action: IDLE
[19:25:10][D][climate:408]:   Current Temperature: nan°C
[19:25:10][D][climate:414]:   Target Temperature: 20.00°C
[19:25:10][D][opentherm_component:095]: update heatingWaterClimate: 0
[19:25:10][D][opentherm_component:096]: update hotWaterClimate: 0
[19:25:12][D][opentherm_component:123]: setBoilerTemperature  at 0.000000 °C (from PID Output)
```
And to be honest I'm nor really sure what to make of it due to lack of experience.

Here's the yaml file I feed to ESPHome: 
```yaml
substitutions:
  devicename: opentherm_basic
  upper_devicename: Opentherm Basic

esphome:
  name: "boiler"
  platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  includes:
    - esphome-opentherm/

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<my key>"

ota:
  password: "<my key>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "<my password>"

captive_portal:

custom_component:
  - lambda: |-
      auto opentherm = new OpenthermComponent();
      return {opentherm};
    components:
      - id: opentherm
output:
  - platform: custom
    type: float
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      auto opentherm_pid_output = new OpenthermFloatOutput();
      openthermComp->set_pid_output(opentherm_pid_output);
      App.register_component(opentherm_pid_output);     
      return {opentherm_pid_output};
    outputs:
      id: pid_output

sensor:
  - platform: custom
    lambda: |-    
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return { 
        openthermComp->boiler_temperature, 
        openthermComp->external_temperature_sensor, 
        openthermComp->return_temperature_sensor, 
        openthermComp->pressure_sensor,
        openthermComp->modulation_sensor,
        openthermComp->heating_target_temperature_sensor
      };
    sensors:
    - name: "Boiler Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "External Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 0  
    - name: "Return Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "Heating Water Pressure"
      unit_of_measurement: hPa
      accuracy_decimals: 1
    - name: "Boiler Modulation"
      unit_of_measurement: "%"
      accuracy_decimals: 0
    - name: "Heating Target Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
      
  - platform: homeassistant
    id: temperature_sensor
    entity_id: sensor.temperature_sensor
    name: "Actual temperature"
    unit_of_measurement: °C
  - platform: template
    id: temperature_sensor_interval
    name: "Actual temperature interval"
    lambda: |-
      return id(temperature_sensor).state;
    update_interval: 1s
  - platform: pid
    name: "PID Climate Result"
    type: RESULT    
  - platform: pid
    name: "PID Climate HEAT"
    type: HEAT    
  - platform: pid
    name: "PID Climate ERROR"
    type: ERROR    

binary_sensor:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {openthermComp->flame};
    binary_sensors:
    - name: "Flame"
      #device_class: heat

switch:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {openthermComp->thermostatSwitch};
    switches:
      name: "Disable PID"
  - platform: template
    name: "PID Climate Autotune"
    turn_on_action:
      - climate.pid.autotune: pid_climate      

climate:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {
        openthermComp->hotWaterClimate, 
        openthermComp->heatingWaterClimate
      };
    climates:
      - id: hot_water
        name: "Hot water"
      - id: heating_water
        name: "Heating water"
  - platform: pid
    id: pid_climate
    name: "PID Climate Controller"
    visual:
      min_temperature: 16 °C
      max_temperature: 28 °C
      temperature_step: 0.5 °C
    sensor: temperature_sensor_interval
    default_target_temperature: 20°C
    heat_output: pid_output
    control_parameters:
      kp: 0.38197
      ki: 0.01012
      kd: 3.60387
```

DIYLESS's page says "polarity does not matter" but just in case it does, here's how I connected mine
B1 -> ground
B2 -> VCC

My boiler is a "Riello Condens" for both central heating and hot water. I checked with my technician and he's positive this thing speaks opentherm.

PINS: I used the included pins to connect the shield to the D1 so they're still be D1 and D2 as per sourcecode.

RP92, 2023-02-04:
The opentherm protocol is working correctly, what didn't work are the entities that can't see the home assistant, try to recreate everything from 0 and try esphome:
   name: $devicename

strk90, 2023-02-20:
> what didn't work are the entities that can't see the home ass

How to solve the problem?

strk90, 2023-02-20:
> 

It doesn't help

strk90, 2023-02-20:
> The opentherm protocol is working correctly, what didn't work are the entities that can't see the home assistant, try to recreate everything from 0 and try esphome: name: $devicename

It doesn't help

strk90, 2023-02-22:
Сan anyone help?

gdschut, 2023-02-22:
If you measure around 36 V I doubt it speaks opentherm... Is this your boiler? https://www.riello.com/international/products/residential-boilers?range=31BB1PEERF? There is nothing about opentherm in the docs and the figure in the installation manual shows on/off thermostat.
If the OpenTherm shield is not connected to opentherm it does not work. Perhaps the boiler can speak opentherm, but only depending on setting or another connector, or perhaps external module required?
OpenTherm connectors are not labeled with VCC or ground, as there is none.

tricker1234, 2023-03-12:
I have the same problem, nothing shows up.

My boiler supports OpenTherm and it worked on another software without ESPhome (DIYLess software only with boiler temp, boiler target temp and current temp).

Has anyone resolved this problem?