rsciriano/ESPHome-OpenTherm #49: Add missing sensors

RP92, 2023-02-04:
There are several sensors that should be added:

1: Hot water outlet temperature
2: Cold water inlet temperature
3: Fan speed
4: pump revolutions
5: date and time present on the card
6: Circulator flow rate
7: DHW flow rate
8: Smoke temperature
9: Maintenance by Displays the number of days by which maintenance must be performed.
10: Some boilers have a domestic hot water micro-storage which can be programmed in time slots (for IMMERGAS boilers this function is called Boost)

Chupaka, 2023-02-06:
> should be added

You mean, could be added? If there are people with devices supporting all of that - they can submit a PR :) My Bosch boiler does not support anything from the list :(

RP92, 2023-02-07:
> 

hi, yes there are many boilers that have many more sensors than your bosch model

en-ver, 2023-02-07:
> > 
> 
> hi, yes there are many boilers that have many more sensors than your bosch model

btw, is there any list of boilers with the most data populated via OT interface? My Buderus Logamax 072-24 is very weak in terms of the data it shares.

Chupaka, 2023-02-08:
> hi, yes there are many boilers that have many more sensors than your bosch model

That's what I'm saying, but what to do with all those non-functioning (in most cases) sensors to avoid questions like "Almost nothing is working, I can only see water temperature, please help me fix that"?

RP92, 2023-02-08:
Just simply esculerdi, whoever has them like me will be able to use them, The code should be ready to work at 360 degrees, then each one on the basis of his boiler enables and disables more or less sensors

Chupaka, 2023-02-08:
Well, looks like an external component form https://github.com/rsciriano/ESPHome-OpenTherm/tree/external-component better fits for this, when you can simply pick all components you need.