rsciriano/ESPHome-OpenTherm #50: ошибка компиляции

ZHNovell, 2023-02-06:
При компиляции выдает ошибку:`INFO Reading configuration /config/esphome/opentherm.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing opentherm (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- OpenTherm Library @ 1.1.3
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/opentherm/.pioenvs/opentherm/src/main.cpp.o
Archiving /data/opentherm/.pioenvs/opentherm/lib67b/libESP8266WiFi.a
Compiling /data/opentherm/.pioenvs/opentherm/libb95/ESPAsyncWebServer-esphome/WebAuthentication.cpp.o
Compiling /data/opentherm/.pioenvs/opentherm/libb95/ESPAsyncWebServer-esphome/WebHandlers.cpp.o
Compiling /data/opentherm/.pioenvs/opentherm/libb95/ESPAsyncWebServer-esphome/WebRequest.cpp.o
Compiling /data/opentherm/.pioenvs/opentherm/libb95/ESPAsyncWebServer-esphome/WebResponses.cpp.o
Compiling /data/opentherm/.pioenvs/opentherm/libb95/ESPAsyncWebServer-esphome/WebServer.cpp.o
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:29:28: error: expected type-specifier before 'OpenthermComponent'
   29 |       auto opentherm = new OpenthermComponent();
      |                            ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:30:24: error: could not convert '{opentherm}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
   30 |       return {opentherm};
      |                        ^
      |                        |
      |                        <brace-enclosed initializer list>
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:37:7: error: 'OpenthermComponent' was not declared in this scope
   37 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |       ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:37:27: error: 'openthermComp' was not declared in this scope; did you mean 'opentherm'?
   37 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                           ^~~~~~~~~~~~~
      |                           opentherm
/config/esphome/opentherm.yaml:37:63: error: expected primary-expression before ')' token
   37 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                                                               ^
/config/esphome/opentherm.yaml:38:39: error: expected type-specifier before 'OpenthermFloatOutput'
   38 |       auto opentherm_pid_output = new OpenthermFloatOutput();
      |                                       ^~~~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:41:35: error: could not convert '{opentherm_pid_output}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::output::FloatOutput*>'
   41 |       return {opentherm_pid_output};
      |                                   ^
      |                                   |
      |                                   <brace-enclosed initializer list>
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:50:7: error: 'OpenthermComponent' was not declared in this scope
   50 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |       ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:50:27: error: 'openthermComp' was not declared in this scope; did you mean 'opentherm'?
   50 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                           ^~~~~~~~~~~~~
      |                           opentherm
/config/esphome/opentherm.yaml:50:63: error: expected primary-expression before ')' token
   50 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                                                               ^
/config/esphome/opentherm.yaml:58:7: error: could not convert '{<expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
   58 |       };
      |       ^
      |       |
      |       <brace-enclosed initializer list>
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:103:7: error: 'OpenthermComponent' was not declared in this scope
  103 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |       ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:103:27: error: 'openthermComp' was not declared in this scope; did you mean 'opentherm'?
  103 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                           ^~~~~~~~~~~~~
      |                           opentherm
/config/esphome/opentherm.yaml:103:63: error: expected primary-expression before ')' token
  103 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                                                               ^
/config/esphome/opentherm.yaml:104:35: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::binary_sensor::BinarySensor*>'
  104 |       return {openthermComp->flame};
      |                                   ^
      |                                   |
      |                                   <brace-enclosed initializer list>
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:112:7: error: 'OpenthermComponent' was not declared in this scope
  112 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |       ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:112:27: error: 'openthermComp' was not declared in this scope; did you mean 'opentherm'?
  112 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                           ^~~~~~~~~~~~~
      |                           opentherm
/config/esphome/opentherm.yaml:112:63: error: expected primary-expression before ')' token
  112 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                                                               ^
/config/esphome/opentherm.yaml:113:46: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::switch_::Switch*>'
  113 |       return {openthermComp->thermostatSwitch};
      |                                              ^
      |                                              |
      |                                              <brace-enclosed initializer list>
/config/esphome/opentherm.yaml: In lambda function:
/config/esphome/opentherm.yaml:124:7: error: 'OpenthermComponent' was not declared in this scope
  124 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |       ^~~~~~~~~~~~~~~~~~
/config/esphome/opentherm.yaml:124:27: error: 'openthermComp' was not declared in this scope; did you mean 'opentherm'?
  124 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                           ^~~~~~~~~~~~~
      |                           opentherm
/config/esphome/opentherm.yaml:124:63: error: expected primary-expression before ')' token
  124 |       OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      |                                                               ^
/config/esphome/opentherm.yaml:128:7: error: could not convert '{<expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::climate::Climate*>'
  128 |       };
      |       ^
      |       |
      |       <brace-enclosed initializer list>
Compiling /data/opentherm/.pioenvs/opentherm/libecb/DNSServer/DNSServer.cpp.o
*** [/data/opentherm/.pioenvs/opentherm/src/main.cpp.o] Error 1
========================= [FAILED] Took 16.47 seconds =========================`

в custom_component:
  - lambda: |-
      auto opentherm = new OpenthermComponent();
      return {opentherm};
    components:
      - id: opentherm    какой ID указывать? я просто не пойму) извините. Остальные ошибки при компиляции из-за этого? Спасибо.

ZHNovell, 2023-02-06:
Извините)
`substitutions:
  devicename: opentherm_basic
  upper_devicename: Opentherm Basic

esphome:
  name: $devicename
  platformio_options:
    lib_deps: 
    - ihormelnyk/OpenTherm Library @ 1.1.3
  includes:
    - esphome-opentherm/

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: INTERNET
  password: "********"
  ap:
    ssid: "${devicename} Fallback"
    password: "********"

captive_portal:

custom_component:
  - lambda: |-
      auto opentherm = new OpenthermComponent();
      return {opentherm};
    components:
      - id: opentherm
output:
  - platform: custom
    type: float
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      auto opentherm_pid_output = new OpenthermFloatOutput();
      openthermComp->set_pid_output(opentherm_pid_output);
      App.register_component(opentherm_pid_output);     
      return {opentherm_pid_output};
    outputs:
      id: pid_output

sensor:
  - platform: custom
    lambda: |-    
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return { 
        openthermComp->boiler_temperature, 
        openthermComp->external_temperature_sensor, 
        openthermComp->return_temperature_sensor, 
        openthermComp->pressure_sensor,
        openthermComp->modulation_sensor,
        openthermComp->heating_target_temperature_sensor
      };
    sensors:
    - name: "Boiler Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "External Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 0  
    - name: "Return Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
    - name: "Heating Water Pressure"
      unit_of_measurement: hPa
      accuracy_decimals: 1
    - name: "Boiler Modulation"
      unit_of_measurement: "%"
      accuracy_decimals: 0
    - name: "Heating Target Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
      
  - platform: homeassistant
    id: temperature_sensor
    entity_id: sensor.temperature_sensor
    name: "Actual temperature"
    unit_of_measurement: °C
  - platform: template
    id: temperature_sensor_interval
    name: "Actual temperature interval"
    lambda: |-
      return id(temperature_sensor).state;
    update_interval: 1s
  - platform: pid
    name: "PID Climate Result"
    type: RESULT    
  - platform: pid
    name: "PID Climate HEAT"
    type: HEAT    
  - platform: pid
    name: "PID Climate ERROR"
    type: ERROR    

binary_sensor:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {openthermComp->flame};
    binary_sensors:
    - name: "Flame"
      #device_class: heat

switch:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {openthermComp->thermostatSwitch};
    switches:
      name: "Disable PID"
  - platform: template
    name: "PID Climate Autotune"
    turn_on_action:
      - climate.pid.autotune: pid_climate      

climate:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {
        openthermComp->hotWaterClimate, 
        openthermComp->heatingWaterClimate
      };
    climates:
      - id: hot_water
        name: "Hot water"
      - id: heating_water
        name: "Heating water"
  - platform: pid
    id: pid_climate
    name: "PID Climate Controller"
    visual:
      min_temperature: 16 °C
      max_temperature: 28 °C
      temperature_step: 0.5 °C
    sensor: temperature_sensor_interval
    default_target_temperature: 20°C
    heat_output: pid_output
    control_parameters:
      kp: 0.38197
      ki: 0.01012
      kd: 3.60387`

ZHNovell, 2023-02-06:
блин да что-
то с форматированием у меня проблема)

ZHNovell, 2023-02-06:
Извините [Chupaka](https://github.com/Chupaka)

ZHNovell, 2023-02-06:
[opentherm.zip](https://github.com/rsciriano/ESPHome-OpenTherm/files/10667964/opentherm.zip)

ZHNovell, 2023-02-06:
Расположение файлов: 
[image]
[image]

Chupaka, 2023-02-06:
Я склонировал содержимое этой репы, скачал и распаковал поверху ваш архив с opentherm.yaml, запустил `docker run --rm -v "${PWD}":/config esphome/esphome compile opentherm.yaml` - и всё отлично собралось. Чтобы не было warning'ов про ICACHE_RAM_ATTR, надо версию `ihormelnyk/OpenTherm Library` поднять с 1.1.3 до 1.1.4 - я там их пофиксил.

Chupaka, 2023-02-06:
Попробовал собрать на 2022.8.0 - тоже всё хорошо собралось. У вас какая версия ESPHome?

ZHNovell, 2023-02-06:
сейчас гляну

ZHNovell, 2023-02-06:
Павел пожалуйста, можете скомпилировать файл? а то я уже и библиотеку 1.1.4 указал и на void IRAM_ATTR поменял, та же ошибка. 
ЯМЛ прикрепил с актуальными данными.
[opentherm.zip](https://github.com/rsciriano/ESPHome-OpenTherm/files/10668268/opentherm.zip)

Chupaka, 2023-02-06:
Так вы docker установите - и компилируйте той командой, что я написал...

Вот папка с итоговыми файлами, но не знаю, чем она вам поможет. Лучше научиться собирать самому.
[opentherm_basic.tar.gz](https://github.com/rsciriano/ESPHome-OpenTherm/files/10668764/opentherm_basic.tar.gz)

Chupaka, 2023-02-08:
@ZHNovell если вопрос снят - закройте, пожалуйста, issue.

closed 2023-02-18 by ZHNovell