rsciriano/ESPHome-OpenTherm #53: Feature Request - Opentherm GATEWAY functionality

faxbio, 2023-02-22:
Hi, based on the same library it seems to be possible to manage also the Gateway Functionality.
The hw needed is available from [Diyless](https://diyless.com/blog/opentherm-gateway-sample) .

The arduino code is [available](https://github.com/diyless/opentherm-gateway-simple), but non integrated in HA. It would be very nice to import the functionality in ESPHOME environment but i'm not able to write the code.

The main advantage is that instead of "sobstitute" the actual thermostats you can keep them working and just override the command when needed.

Can you try to support it?

Mio3000, 2023-02-23:
Following this request, im interested also for this functionality

pimw1, 2023-05-13:
Me as well.

sakrut, 2023-08-31:
Hi :D 
@faxbio @Mio3000  @pimw1 
I made some changes on my fork and i use this in my hose with gateway bord.
https://github.com/sakrut/ESPHome-OpenTherm-Gateway

Unfortunatly currently only to read data, and without fit to new branch(an external component)
If i will sort out problems with load and inject set points then i will try to prepare pr's

and here is what is required to use this commponent as gateway where "processRequest" is most important
https://github.com/sakrut/ESPHome-OpenTherm-Gateway/commit/ce527930c5f24a71f512bc8c36b26b8a006c166e

faxbio, 2023-10-22:
@sakrut Any news? Did you manage to override set point temperature?

Mio3000, 2023-12-01:
@sakrut thx for great work, i see there the dallas sensor, pls do i need it to have this working? or can i use some thermometer from HA?

sakrut, 2023-12-01:
@Mio3000  you can remove dallas sensor, i use this [hadrware ](https://diyless.com/product/esp8266-opentherm-gateway) where this sensor is included.
And thermometer  from HA is not required, becouse i deleted PID. the main controler manage this so work with pid is not nessery

Mio3000, 2023-12-01:
thank you, looks good, and it would be possible also add diagnostic_event and fault_indication? and also ch_active with dhw_active? just as binary sensors
or i can add them somehow here:
```
binary_sensor:
  - platform: custom
    lambda: |-
      OpenthermComponent *openthermComp = (OpenthermComponent*) id(opentherm);
      return {openthermComp->flame};
    binary_sensors:
    - name: "Flame"
      #device_class: heat

```