rsciriano/ESPHome-OpenTherm #54: OpenTherm.h: No such file or directory

tricker1234, 2023-03-12:
Hi group!
Had someone an error during instalation?

"In file included from src/main.cpp:51:src/opentherm_component.h:3:10: fatal error: OpenTherm.h: No such file or directory"

I copied 0.5 version to config->esphome-> (there is opentherm.yaml and esphome-opentherm)

Chupaka, 2023-03-12:
Hi.
OpenTherm.h is external dependency. From `opentherm.yaml`:
```
esphome:
  platformio_options:
    lib_deps:
    - ihormelnyk/OpenTherm Library @ 1.1.4
```
So it's PlatformIO who should provide OpenTherm.h to esphome during compilation...

Any other logs of your compilation process?

tricker1234, 2023-03-12:
Here you have screen from compilation

[image]

tricker1234, 2023-03-12:
I have read some topics and found solution: uninstalled ESPhome and installed ESPhome dev version