rsciriano/ESPHome-OpenTherm #57: Viessmann - long time for an operation

pruwait, 2023-08-06:
[14:14:12][W][component:204]: Component custom_component took a long time for an operation (2.31 s).
[14:14:12][W][component:205]: Components should block for at most 20-30ms

How can I fix it?

OpenthermComponent(): PollingComponent(3000) > OpenthermComponent(): PollingComponent(10000)  is not working

Chupaka, 2023-08-06:
It's because the component uses synchronous mode of operation via OpenTherm Protocol. You can't do anything with this, I believe, just ignore for now.

closed 2023-08-08 by pruwait

mool, 2023-08-23:
I have the same warnings, and it looks like as time passes it starts to work with more delay and I have to restart it to make it respond when it is supposed to.