rsciriano/ESPHome-OpenTherm #64: Custom component took a long time

Alfagek, 2024-02-10:
Hi All,

I’m using the diyless master thermostat shield with a wmos D1. And esphome 2023.12.9. Also did try with opentherm library 1.1.5 instead of 1.1.4. But same result. 

It’s connect to a 2.4ghz WiFi and the OTG heater is a AWB Thermomaster 3HR

But i get some sensors from OTG that are sometime “unavailable” for a few seconds and then it is back again. 

This error i see in my esphome log from the device on the moments when a sensor is NAN. Sometime it’s the flame then it’s the return temperature or the modulation. 

14:59:32][W][component:214]: Component custom_component took a long time for an operation (2.75 s).
[14:59:32][W][component:215]: Components should block for at most 20-30ms.

This firmware does work, only this “annoying” NAN sensors. So keep up the good work 👍🏻

I hope someone has a solution to fix this “problem”. 

Thanks in advanced!

VladaDHS, 2024-02-10:
Hi, 
based on the update time of the projekt loke like a dead end :(

Alfagek, 2024-02-10:
> Hi, based on the update time of the projekt loke like a dead end :(

code owner did made changes 4 months ago. So there is some life. And why change if it’s functional

VladaDHS, 2024-02-10:
On main bracnch yes. I miss that. 
But for me even main version is not working :/ device srtay in boot loop for some reason.